PX4/PX4-Autopilot #7364: Snapdragon - krait update param fail

ChristophTobler, 2017-06-06:
Tested with https://github.com/PX4/Firmware/pull/7359

The output is http://pasted.co/12c48f79

dagar, 2017-06-06:
Is this from the param struct being out of sync?

ChristophTobler, 2017-06-06:
I though this happens if the Linux and DSP do not have the same parameters. @jywilson could you have a quick look?

dagar, 2017-06-06:
@jywilson to jump straight in, the question is if this branch works https://github.com/PX4/Firmware/pull/7359

jywilson, 2017-06-06:
It would help if you could provide the debug messages generated by the DSP through minidm.  It looks like the communication to the aDSP is not working for some reason and the minidm log would help determine the reason for the communication issue..  

The most common reason for this is that a function being called by the AppsProc is not available in the .so files running on the aDSP.  This looks like a communication problem, but is really a broken dynamic link.  Since the Hexagon SDK linker assumes all unresolved symbols for aDSP .so files are dynamic links (instead of showing the as "unresolved external symbol" linker errors), this problem is not uncommon.  The minidm logs will show this problem clearly.

If the aDSP image is restarting due to a runtime exception, this same symptom will appear.  The minidm logs are not as helpful for this type of problem, since the restart causes the last few log messages before the exception to be dropped from the aDSP buffer before they are output to minidm.  We can at least identify the restart condition from the minidm logs.

Jim W.

ChristophTobler, 2017-06-07:
Thanks @jywilson 
Sorry, I forgot about the minidm... here it is
```
Running mini-dm version: 3.0
Device found with Product ID 0x9025. Continuing...
mini-dm is waiting for a DMSS connection...
DMSS is connected. Running mini-dm...
[08500/03]  04:30.360  HAP:58:undefined PLT symbol _ZN7control10BlockParamIiEC1EPNS_5BlockEPKcb (743) /libpx4.so  0303  symbol.c
[08500/03]  04:30.385  HAP:58:undefined PLT symbol _ZN7control10BlockParamIiEC1EPNS_5BlockEPKcb (743) /libpx4.so  0303  symbol.c
[08500/03]  04:30.415  HAP:58:undefined PLT symbol _ZN7control10BlockParamIiEC1EPNS_5BlockEPKcb (743) /libpx4.so  0303  symbol.c
[08500/03]  04:30.443  HAP:58:undefined PLT symbol _ZN7control10BlockParamIiEC1EPNS_5BlockEPKcb (743) /libpx4.so  0303  symbol.c
[08500/03]  04:30.471  HAP:58:undefined PLT symbol _ZN7control10BlockParamIiEC1EPNS_5BlockEPKcb (743) /libpx4.so  0303  symbol.c
[08500/03]  04:30.495  HAP:58:undefined PLT symbol _ZN7control10BlockParamIiEC1EPNS_5BlockEPKcb (743) /libpx4.so  0303  symbol.c
[08500/03]  04:30.525  HAP:58:undefined PLT symbol _ZN7control10BlockParamIiEC1EPNS_5BlockEPKcb (743) /libpx4.so  0303  symbol.c
[08500/03]  04:30.550  HAP:58:undefined PLT symbol _ZN7control10BlockParamIiEC1EPNS_5BlockEPKcb (743) /libpx4.so  0303  symbol.c
[08500/03]  04:30.573  HAP:58:undefined PLT symbol _ZN7control10BlockParamIiEC1EPNS_5BlockEPKcb (743) /libpx4.so  0303  symbol.c
[08500/03]  04:30.595  HAP:58:undefined PLT symbol _ZN7control10BlockParamIiEC1EPNS_5BlockEPKcb (743) /libpx4.so  0303  symbol.c
[08500/03]  04:30.622  HAP:58:undefined PLT symbol _ZN7control10BlockParamIiEC1EPNS_5BlockEPKcb (743) /libpx4.so  0303  symbol.c
[08500/03]  04:30.648  HAP:58:undefined PLT symbol _ZN7control10BlockParamIiEC1EPNS_5BlockEPKcb (743) /libpx4.so  0303  symbol.c
[08500/03]  04:30.673  HAP:56:undefined PLT symbol _ZN7control10BlockParamIiEC1EPNS_5BlockEPKcb (743) /libpx4.so  0303  symbol.c
[08500/03]  04:30.696  HAP:58:undefined PLT symbol _ZN7control10BlockParamIiEC1EPNS_5BlockEPKcb (743) /libpx4.so  0303  symbol.c
```

jywilson, 2017-06-08:
@ChristophTobler 

The minidm output indicates that a function is referenced in the aDSP code (compiled into dynamically loaded .so files) for which there is no actual code.  This will cause the associated .so to fail to load and since most of the flight stack is contained in the libpx4.so, that means that the flight stack will not run.  

The name of the function (demangled from the C++ symbol name above) is: control::BlockParam<int>::BlockParam(control::Block*, char const*, bool)

Jim W.

dagar, 2017-06-08:
Fixed here - https://github.com/PX4/Firmware/pull/7359/commits/343578eaa1e4631bb5cb3e68b705345d1dce0fdb
Apparently int32_t != int here.

jywilson, 2017-06-08:
Good catch!

ChristophTobler, 2017-06-12:
closing this as it has been fixed in the PR and it doesn't affect master

closed 2017-06-12 by ChristophTobler