PX4/PX4-Autopilot #7365: X2.1 Mag Problem: The MAG0 is 180 deg offset.

tubeme, 2017-06-06:
Hello guys,

We spent a couple of days researching on the Magnetometer behavior of the PX4 with a different Hardware because we are moving toward using 1.6.x version of the firmware.

Lastly we installed the 1.6.1 on a X2.1 board and found that our Mag0 is a 180 degrees offset from the correct direction it was pointed at. The internal mag was fine only the external MAG0 had this offset. Later we found this https://github.com/PX4/Firmware/issues/7358 and we experience exactly the same problem despite we have a different hardware.

There were was only one solution to the problem we found so far. Just phisicaly rotate the external mag, ie the GPS pointing in the opposite direction. We did two tests. 

1. Mount the GPS+MAG in the correct direction and then do the Mag Calibration and then after the calibration was OK rotate the GPSin the opposite direction.

2. Mount the GPS+MAG in the opposite direction then do the Mag Calibration and leave the GPS mag at its current 180 position. 

In both ways the results are the same. The internal and the external mags were in line. So there was no change if the external mag was disconnected ie failing. So given we mount the external mag 180 degrees everything was OK, we could even try to fly it and most likely we will have a successful flights.

Besides this obvious problem we continued doing our Magnetometer tests because we've had another problems with our X2.1 board a month ago. when we started flying with it. Because of the bootloader (v5) in the X2.1 it was recognized as ID:9 then we could flash the 1.5.0 RC1 on it with a QGC 3.1.3 and then we've had some strange thing happen. The Gyros and the Mag had a 90 degree offset. Nothing was correct. So with some tests we used a 90degrees offset for the controller and everything started working OK in terms of the mag and gyros. We could see a correct movements in the Horizon and in the MAG. We've had some successful flights in Manual.

Just recently mr.Arsov  updated the X2.1 bootloader to a newer version and with the current master of QGC the board is correctly recognized as ID 33 and then we were able to flash the correct FW which is AUAV-X2.1. And once flashed we've got this 180degree offset. We could not flash anymore the 1.5.0 RC1 because the QGC prevented this. Now the QGC is checking on the board ID and if it is not a match it does not allow flashing. So we could not try this scenario.

So now we continued on our tests. Our benchmark system is X2+PX4 1.5.0RC1. This version is the most reliable fro us so far. We fly it for 9 months very intensively. It performs perfect. We did not have a single crash with it. We fly it in Manual, Asisted modes as well. We did hundreds of missions on a different hardware. It does not matter if we have some unbalances in the frame or uncalibrated ESCs, some different motors some aerodynamic changes, it just does the best flying so far. So we compare everything with this benchmark system.

So we continued on the Magnetometer tests for the last couple of days to see if there are any differences between the 1.5.0 RC1 and 1.6.1 behavior. We used broad range of hardware in the tests, different controllers Pixhawk original, HKPilot32, X2 and also a couple of different GPS+Mag combos that we fly on a regular basis. We had Quanum LEA-6H gps, UBlox NEO-M8N and a small UBlox 7 Nano GPS. We found a pattern in the whole thing and we want to show you in two videos what the differences are. For the demonstration we use our bench test system which is HKPilot32 + Quanum LEA-6H GPS+MAG. The only change in the videos is the version of the FW. We cannot provide Log data because we did not arm the machines, we just wanted to explore the Mag behavior.  Here is what are the differences in two Videos:

1.5.0 RC
https://youtu.be/38KQdxaMqWY

1.6.1
https://youtu.be/KNG7h6wopgw

We don't know if this is designed behavior or it is misbehavior but there is significant difference.

MAG0 (the external one) is very very slow and lagy in 1.6.1 and in case of failure the change is taking up to a minute or two to get the internal mags the correct heading. It seems to us that the 1.5.0 RC1 has very good way of moving between the two mags in case of failure. If we imagine a flight this minute or two in 1.6.1 could be fatal. 

What do you think?

mhkabir, 2017-06-06:
Can you give me a parameter dump from both working and non-working cases please?

mhkabir, 2017-06-06:
> Lastly we installed the 1.6.1 on a X2.1 board and found that our Mag0 is a 180 degrees offset from the correct direction it was pointed at. The internal mag was fine only the external MAG0 had this offset.

What is CAL_MAG0_ROT, and what is the actual orientation of the autopilot and external mag with respect to the airframe.

Regarding your videos - the conclusions you draw are flawed. If you have a failure in-flight, the heading reported by all the mags are the same. You would not see the slow convergence you talk about, because there would never be such a large heading difference between the failed sensor and the one we switch over to. 

The slower convergence on 1.6 is by design, as you can see that the EKF rejects step changes in heading (which is inconsistent with the gyro/accel rotation estimate) which prevents weird magnetic field variations from destroying the attitude estimate.

mhkabir, 2017-06-06:
Fix is here : https://github.com/PX4/Firmware/issues/7366

closed 2017-06-06 by mhkabir