PX4/PX4-Autopilot #7368: Feature: support for GPS_GLOBAL_ORIGIN

TSC21, 2017-06-06:
I'm planning to add support for `GPS_GLOBAL_ORIGIN` (and `SET_GPS_GLOBAL_ORIGIN` which can be triggered by a cmd for example), and consequently add support for `LOCAL_POSITION_NED_SYSTEM_GLOBAL_OFFSET` as the offset between the `global_origin` and the `local_position`.

My idea is to implement similar to `HOME_POSITION` but, in terms of usage we can think of something more clear. The idea will be allow the FCU to be aware of the offset between the local frame and the global/earth frame as it moves and, in situations the GPS signal is denied or gets lost, the vehicle is still able to navigate, acquire landmarks or even get in and out of buildings, having always a relation between the frames. This can also be useful for multi-system interaction on this kind of environments.

Any suggestions are welcomed. A feature list with them would be helpful before I start digging and coding only with my concept.

mhkabir, 2017-06-07:
> offset between the global_origin and the local_position.

I don't understand this part.

TSC21, 2017-06-07:
Sorry, I meant `local_origin`.

TSC21, 2017-06-07:
Probably I can sync this with https://github.com/PX4/Firmware/tree/global_to_local. Even if we consider that the local and global origins are coincident, in multi-entities systems we have specific local frames to each entity while we can define a common global origin to all.
@ChristophTobler what do you think?

ChristophTobler, 2017-06-07:
I think it's generally a good idea but I'm not sure this is needed with the change from global to local coordinates we are currently doing. With these changes it will be possible to lose GPS and rely on e.g. vision during the flight without failing since the estimator works in a local frame and also publishes in local coordinates (also the position controller).
Or is there a special case which I am not thinking of?

TSC21, 2017-06-07:
You are right and that was something that Kabir confirmed me. The special case could be, as I made reference, multi-entity systems (like swarms) where the local coordinates are specific to each but the global origin is the same.
Mavros already supports this in its side so this would be a straightforward implementation. Just wanted to receive some feedback on other features we can push.

TSC21, 2017-06-07:
@ChristophTobler are you good with syncing with https://github.com/PX4/Firmware/tree/global_to_local and add a possibility of externally define a `global_origin` plus output the offset through `LOCAL_POSITION_NED_SYSTEM_GLOBAL_OFFSET` if it is not coincident with the `local_origin`? I make a PR against that branch so you can evaluate it.

ChristophTobler, 2017-06-14:
True! I guess that would make sense.

TSC21, 2017-06-15:
> True! I guess that would make sense.

OK. I'm going to start the implementation and issue a PR as soon as I can. Thanks

TSC21, 2018-02-21:
I would still like to make this happen.

GigaFlopsis, 2018-05-31:
This method will be very useful for control of the swarm. I do not quite understand, is it in the firmware? if so, how do I set the parameter via mavros?

TSC21, 2018-05-31:
> This method will be very useful for control of the swarm. I do not quite understand, is it in the firmware? if so, how do I set the parameter via mavros?

This is not supported yet.

GigaFlopsis, 2018-05-31:
Do you need help to realize this?
Now i develop a separate local position relative to the global one with the help of geolibs in ros. It is now i developed a separate local position relative to the global one with the help of geolibs in ros. Now this is an extra setup.

BazookaJoe1900, 2018-11-18:
I think that the current implementation has been wronglly set the origin with message 'GPS_GLOBAL_ORIGIN' instead of using 'SET_GPS_GLOBAL_ORIGIN'.
The GPS_GLOBAL_ORIGIN stream is not implemented as far as I see, where can I find this delta
@TSC21 does this feature still relevant for you?

TSC21, 2018-11-18:
@BazookaJoe1900 it's relevant but I am currently not working in any project that support this. And yes the added support was to `GPS_GLOBAL_ORIGIN` and should have been to `SET_GPS_GLOBAL_ORIGIN` (https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_receiver.cpp#L1175). Though what this does is setting the origin for the projection between the geographic coordinates and the local coordinates, which is not what I was actually aiming for on this feature, as you can read on the above comments. If you want to check the message definitions, you have them here: https://github.com/PX4/ecl/blob/master/geo/geo.h.

TSC21, 2019-08-12:
Still relevant

hamishwillee, 2019-08-13:
Note, there is "actual" work going on in this in #12665. @TSC21 can you engage with that?

mrpollo, 2019-11-20:
Ping @TSC21

Please provide an update.

TSC21, 2019-11-22:
@mrpollo I have to allocate some time for this first. I will let the interested parts know when this is going to be rebooted. This will be for sure in time for the 1.11 release.

junwoo091400, 2023-03-01:
Judging from https://github.com/PX4/PX4-Autopilot/pull/21219, it seems that this feature was already implemented & exists?