PX4/PX4-Autopilot #7376: Parameter save nearly taking 500ms to complete

NaterGator, 2017-06-07:
On pixhawk the param save process is getting dangerously close to 500ms. Once it crosses this threshold it will trigger an FMU timeout on the IO and put the servos into failsafe positions for a brief bit. 

Not yet sure where the hottest spots are in the param save code, but likely candidates seem to by BSON encoding (especially float to double conversion) and SPI locking for the ramtron device.

LorenzMeier, 2017-06-07:
It should not lock out scheduling for that time frame but only that single bus, which has no sensors. If that's not the case then we need to check what is causing the incorrect behavior.

NaterGator, 2017-06-07:
Lorenz, I tried extending the sleep time [here](https://github.com/PX4/Firmware/blob/e0d81b4c5e0f98d0e0f031242057e5497930213f/src/modules/systemlib/param/param.c#L1093) but it doesn't seem to have an effect on the IO timeout. I suppose it is possible there is a param write lock being waited on that is causing a priority inversion, though I'm not sure what task would be routinely acquiring the param write lock.

bkueng, 2017-06-08:
> I suppose it is possible there is a param write lock being waited on that is causing a priority inversion, though I'm not sure what task would be routinely acquiring the param write lock.

If `param_get` or `param_set` is called during a save, it would block. But none of these should be called regularly, and last time I checked, there wasn't.

LorenzMeier, 2017-08-20:
We fixed priorities since - could you re-test?

NaterGator, 2017-08-22:
I can confirm this is fixed; the px4io doesn't timeout when performing a long / expensive param save operations. 

Having said that the cost of storing a float param is about 10ms _per parameter_. The cost for an int32 param isn't perceptible. I suspect that is due to the float <-> double conversion, not sure if it something we care about tracking or addressing.

dagar, 2017-08-22:
10 milliseconds per float parameter save? That seems insane.

Correct me if I'm wrong, but does the bson type even matter? I think we only need to catch type changes on import. What about the byte array type for floats instead of double?

There's 80-90 float params for typical sensor calibration, then potentially another 150 if you're using temperature compensation. Overall parameters are > 75% float.

NaterGator, 2017-08-22:
It seems insane to me too which is why I mentioned it. I can't comment on how that performance scales with respect to the total number of parameters but a default fixedwing system was saving params in 690ms, I added 32 useless float params and observed average save time increase to 1013ms. 

I dug into the tinyBSON source a bit and it did not strike me as particularly important that we store doubles instead of floats except that doubles are defined in the spec and floats are not. I guess my primary concern is with respect to breaking anyone who may be using file-based parameter storage and moving it around or inspecting it with other BSON tools. I don't know if that is a use-case out there... but they may well prefer we not break the BSON spec. If that were the case it may make sense to special-case the FRAM / flash based solutions that are exclusive to microprocessor systems since they aren't going to be exporting BSON files anyways. 

Using BSON byte arrays may work, but presumably we'd have to have some sort of marker to distinguish floats from any other valid use of BSON byte arrays.

dagar, 2017-08-22:
I think we should do a quick hack to see what the best could be (saving every float as a 4 byte type with no conversion).
If it seems like it's worth the effort then we can consider the tradeoffs of losing BSON compatibility or having floats in byte buffers, etc. I can't imagine people need to inspect/edit param files outside of PX4, but who knows. I don't think anyone uses PARAM_TYPE_STRUCT.

For backwards compatibility I think we're good as long as we keep the double import.

NaterGator, 2017-08-22:
Yeah, it would probably be worthwhile to run this under the profiler to ID hotspots (or instrument parameter codepaths by hand). @davids5 I think I saw some activity around PMP, is it still valid to use?

davids5, 2017-08-22:
@NaterGator - yes with an understanding as described [here](https://github.com/PX4/Firmware/issues/6840#issuecomment-323853061)  

I have PROBES on all the components  of a FMUV2 (no silicon errata)  (FRAM) at the moment so let me know if you need a snap of some timing. 
If you can share a param file with a lot of non defaults I can tell you what is overhead and what is SPI activity.

bkueng, 2018-01-30:
This got fixed in https://github.com/PX4/Firmware/pull/8391. Param saving takes now a few milliseconds.

closed 2018-01-30 by bkueng