PX4/PX4-Autopilot #7378: FW Flashing Stable: wrong reporting.

tubeme, 2017-06-08:
Hello guys,

I'm flashing Stable with QGC. Then it reports it as 1.6.1dev. I see that latest stable is 1.6.2 but there is no binary i can flash. And I'm not sure what Stable is flashing in QGC... Either it is wrongly reported version or Stable flashes something else.

mhkabir, 2017-06-08:
@dagar @LorenzMeier looks like tagging issue again.

dagar, 2017-06-08:
@mhkabir something else is going on here. If you fetch everything the v1.6.2 tag is there, but not showing in the stable branch.

	50ecee9 (HEAD -> stable, px4/stable) FINALLY fix mag rotation issues. (#7366)
	b4b8a03 (tag: v1.6.1) mavlink: new MAV_CMD_IMAGE_START_CAPTURE spec
	ab14de9 sensors : correctly handle internal magnetometer rotations
	8ff7346 BMI055: Ensure the accel driver sets the device ID in the report
	3c1af57 BMI055: Ensure gyro device ID gets copied to report
	d49f1be FMUv4 config: Ensure enough stack space in user main function
	cc3d287 uORB listener: Expand command to allow selecting the instance to print
	03a762f commander: arming with critical battery is not allowed
	3b85ac0 Multicopter mixer: When saturating, compute the new yaw value based on the reduced thrust value
	29b0043 (tag: v1.6.0) Update README
	a9ad826 Commander: Relax gyro consistency check as users without temperature compensation do suffer from it.
	ed5cf9f (tag: v1.6.0-rc4) commander : simplify platform defines

dagar, 2017-06-08:
The commit was tagged in master. If you cherry-pick to another branch the tag doesn't follow. Git can follow merges back for versioning, but that doesn't help in this case.

[image]

dagar, 2017-12-06:
Should be resolved.

closed 2017-12-06 by dagar