PX4/PX4-Autopilot #7385: Initialize Error calling the uorb fastrpc initalize method..

Seanmatthews, 2017-06-08:
I switched up to 3.1.3 Snapdragon Flight image  and v1.6.2 of PX4. Built PX4 with `make eagle_legacy_default`. What's going on here?

Mind the large error dump:
> px4 starting.
> 
> ERROR [muorb] Initialize Error calling the uorb fastrpc initalize method..
> INFO  [logger] logger started (mode=all)
> ERROR [param] Parameter MAV_BROADCAST not found.
> ERROR [param] Parameter SDLOG_PRIO_BOOST not found.
> ERROR [muorb] ERROR: FastRpcWrapper Not Initialized
> ERROR [lib__controllib] error finding param: GF_ACTION
> ERROR [lib__controllib] error finding param: GF_ALTMODE
> ERROR [lib__controllib] error finding param: GF_SOURCE
> ERROR [lib__controllib] error finding param: GF_COUNT
> ERROR [lib__controllib] error finding param: GF_MAX_HOR_DIST
> ERROR [lib__controllib] error finding param: GF_MAX_VER_DIST
> ERROR [lib__controllib] error finding param: MIS_LTRMIN_ALT
> ERROR [lib__controllib] error finding param: MIS_YAW_TMT
> ERROR [lib__controllib] error finding param: MIS_YAW_ERR
> ERROR [lib__controllib] error finding param: VT_WV_LND_EN
> ERROR [lib__controllib] error finding param: VT_WV_TKO_EN
> ERROR [lib__controllib] error finding param: VT_WV_LTR_EN
> ERROR [lib__controllib] error finding param: NAV_FORCE_VT
> ERROR [lib__controllib] error finding param: VT_B_TRANS_DUR
> ERROR [lib__controllib] error finding param: MIS_ONBOARD_EN
> ERROR [lib__controllib] error finding param: MIS_TAKEOFF_ALT
> ERROR [lib__controllib] error finding param: MIS_DIST_1WP
> ERROR [lib__controllib] error finding param: MIS_ALTMODE
> ERROR [lib__controllib] error finding param: MIS_YAWMODE
> ERROR [lib__controllib] error finding param: NAV_FORCE_VT
> ERROR [lib__controllib] error finding param: FW_CLMBOUT_DIFF
> ERROR [lib__controllib] error finding param: MIS_LTRMIN_ALT
> ERROR [lib__controllib] error finding param: MIS_YAW_TMT
> ERROR [lib__controllib] error finding param: MIS_YAW_ERR
> ERROR [lib__controllib] error finding param: VT_WV_LND_EN
> ERROR [lib__controllib] error finding param: VT_WV_TKO_EN
> ERROR [lib__controllib] error finding param: VT_WV_LTR_EN
> ERROR [lib__controllib] error finding param: NAV_FORCE_VT
> ERROR [lib__controllib] error finding param: VT_B_TRANS_DUR
> ERROR [lib__controllib] error finding param: MIS_LTRMIN_ALT
> ERROR [lib__controllib] error finding param: MIS_YAWMODE
> ERROR [lib__controllib] error finding param: MIS_LTRMIN_ALT
> ERROR [lib__controllib] error finding param: MIS_YAW_TMT
> ERROR [lib__controllib] error finding param: MIS_YAW_ERR
> ERROR [lib__controllib] error finding param: VT_WV_LND_EN
> ERROR [lib__controllib] error finding param: VT_WV_TKO_EN
> ERROR [lib__controllib] error finding param: VT_WV_LTR_EN
> ERROR [lib__controllib] error finding param: NAV_FORCE_VT
> ERROR [lib__controllib] error finding param: VT_B_TRANS_DUR
> ERROR [lib__controllib] error finding param: MIS_TAKEOFF_ALT
> ERROR [lib__controllib] error finding param: MIS_LTRMIN_ALT
> ERROR [lib__controllib] error finding param: MIS_YAW_TMT
> ERROR [lib__controllib] error finding param: MIS_YAW_ERR
> ERROR [lib__controllib] error finding param: VT_WV_LND_EN
> ERROR [lib__controllib] error finding param: VT_WV_TKO_EN
> ERROR [lib__controllib] error finding param: VT_WV_LTR_EN
> ERROR [lib__controllib] error finding param: NAV_FORCE_VT
> ERROR [lib__controllib] error finding param: VT_B_TRANS_DUR
> ERROR [lib__controllib] error finding param: MIS_LTRMIN_ALT
> ERROR [lib__controllib] error finding param: MIS_YAW_TMT
> ERROR [lib__controllib] error finding param: MIS_YAW_ERR
> ERROR [lib__controllib] error finding param: VT_WV_LND_EN
> ERROR [lib__controllib] error finding param: VT_WV_TKO_EN
> ERROR [lib__controllib] error finding param: VT_WV_LTR_EN
> ERROR [lib__controllib] error finding param: NAV_FORCE_VT
> ERROR [lib__controllib] error finding param: VT_B_TRANS_DUR
> ERROR [lib__controllib] error finding param: RTL_RETURN_ALT
> ERROR [lib__controllib] error finding param: MIS_LTRMIN_ALT
> ERROR [lib__controllib] error finding param: RTL_DESCEND_ALT
> ERROR [lib__controllib] error finding param: RTL_LAND_DELAY
> ERROR [lib__controllib] error finding param: RTL_MIN_DIST
> ERROR [lib__controllib] error finding param: MIS_LTRMIN_ALT
> ERROR [lib__controllib] error finding param: MIS_YAW_TMT
> ERROR [lib__controllib] error finding param: MIS_YAW_ERR
> ERROR [lib__controllib] error finding param: VT_WV_LND_EN
> ERROR [lib__controllib] error finding param: VT_WV_TKO_EN
> ERROR [lib__controllib] error finding param: VT_WV_LTR_EN
> ERROR [lib__controllib] error finding param: NAV_FORCE_VT
> ERROR [lib__controllib] error finding param: VT_B_TRANS_DUR
> ERROR [lib__controllib] error finding param: NAV_RCL_LT
> ERROR [lib__controllib] error finding param: MIS_LTRMIN_ALT
> ERROR [lib__controllib] error finding param: MIS_YAW_TMT
> ERROR [lib__controllib] error finding param: MIS_YAW_ERR
> ERROR [lib__controllib] error finding param: VT_WV_LND_EN
> ERROR [lib__controllib] error finding param: VT_WV_TKO_EN
> ERROR [lib__controllib] error finding param: VT_WV_LTR_EN
> ERROR [lib__controllib] error finding param: NAV_FORCE_VT
> ERROR [lib__controllib] error finding param: VT_B_TRANS_DUR
> ERROR [lib__controllib] error finding param: NAV_DLL_CH_T
> ERROR [lib__controllib] error finding param: NAV_DLL_CH_LAT
> ERROR [lib__controllib] error finding param: NAV_DLL_CH_LON
> ERROR [lib__controllib] error finding param: NAV_DLL_CH_ALT
> ERROR [lib__controllib] error finding param: NAV_AH_LAT
> ERROR [lib__controllib] error finding param: NAV_AH_LON
> ERROR [lib__controllib] error finding param: NAV_AH_ALT
> ERROR [lib__controllib] error finding param: NAV_DLL_AH_T
> ERROR [lib__controllib] error finding param: NAV_DLL_N
> ERROR [lib__controllib] error finding param: NAV_DLL_CHSK
> ERROR [lib__controllib] error finding param: MIS_LTRMIN_ALT
> ERROR [lib__controllib] error finding param: MIS_YAW_TMT
> ERROR [lib__controllib] error finding param: MIS_YAW_ERR
> ERROR [lib__controllib] error finding param: VT_WV_LND_EN
> ERROR [lib__controllib] error finding param: VT_WV_TKO_EN
> ERROR [lib__controllib] error finding param: VT_WV_LTR_EN
> ERROR [lib__controllib] error finding param: NAV_FORCE_VT
> ERROR [lib__controllib] error finding param: VT_B_TRANS_DUR
> ERROR [lib__controllib] error finding param: MIS_LTRMIN_ALT
> ERROR [lib__controllib] error finding param: MIS_YAW_TMT
> ERROR [lib__controllib] error finding param: MIS_YAW_ERR
> ERROR [lib__controllib] error finding param: VT_WV_LND_EN
> ERROR [lib__controllib] error finding param: VT_WV_TKO_EN
> ERROR [lib__controllib] error finding param: VT_WV_LTR_EN
> ERROR [lib__controllib] error finding param: NAV_FORCE_VT
> ERROR [lib__controllib] error finding param: VT_B_TRANS_DUR
> ERROR [lib__controllib] error finding param: NAV_GPSF_LT
> ERROR [lib__controllib] error finding param: NAV_GPSF_R
> ERROR [lib__controllib] error finding param: NAV_GPSF_P
> ERROR [lib__controllib] error finding param: NAV_GPSF_TR
> ERROR [lib__controllib] error finding param: MIS_LTRMIN_ALT
> ERROR [lib__controllib] error finding param: MIS_YAW_TMT
> ERROR [lib__controllib] error finding param: MIS_YAW_ERR
> ERROR [lib__controllib] error finding param: VT_WV_LND_EN
> ERROR [lib__controllib] error finding param: VT_WV_TKO_EN
> ERROR [lib__controllib] error finding param: VT_WV_LTR_EN
> ERROR [lib__controllib] error finding param: NAV_FORCE_VT
> ERROR [lib__controllib] error finding param: VT_B_TRANS_DUR
> ERROR [lib__controllib] error finding param: NAV_MIN_FT_HT
> ERROR [lib__controllib] error finding param: NAV_FT_DST
> ERROR [lib__controllib] error finding param: NAV_FT_FS
> ERROR [lib__controllib] error finding param: NAV_FT_RS
> ERROR [lib__controllib] error finding param: MC_YAWRAUTO_MAX
> ERROR [lib__controllib] error finding param: NAV_LOITER_RAD
> ERROR [lib__controllib] error finding param: NAV_ACC_RAD
> ERROR [lib__controllib] error finding param: NAV_FW_ALT_RAD
> ERROR [lib__controllib] error finding param: NAV_MC_ALT_RAD
> INFO  [mavlink] mode: Normal, data rate: 1000000 B/s on udp port 14556 remote port 14550
> WARN  [mavlink] Save params and reboot to change SYSID
> WARN  [mavlink] Save params and reboot to change COMPID
> Sleeping for 1 s; (1000000 us).

dagar, 2017-06-08:
Should be fixed by this PR - https://github.com/PX4/Firmware/pull/7359
This is the relevant commit. https://github.com/PX4/Firmware/pull/7359/commits/62137fd45c22d090b661dc7d319b53458801de7d

Seanmatthews, 2017-06-08:
@dagar Thanks for the reference. That thread adequately explains (and solves) the issue. I'll close this out.

closed 2017-06-08 by Seanmatthews