PX4/PX4-Autopilot #7386: Lack of documentation

sparky002, 2017-06-08:
A few outstanding questions for a fixed wing aircraft with landing gear.

Wheel Steering:
1.) What mode does wheel steering effect while in take-off role on the ground? 
2.) Does wheel steering only effect stabilize, or auto/mission modes, or both. 
3.) Does wheel steering only effect the aircraft while it is on the ground, or in the air as well.
4.) How does the Pixhawk know when to stop using wheel steering if it is disabled after take-off.
5.) What is the relation between wheel steering and non wheel steering yaw control, on the ground, in the air? Which one proceeds the other at what time?
6.) How does an aircraft on the ground in mission or stabilize mode control heading and/or zero rate attitude with wheel steering disabled?
7.) Is it possible to have a specific set of tunable yaw and L1 parameters for "on the ground" control and another for "in flight" control. Reason: yaw control on the ground is drastically different do to friction, weight on the ground, velocity, and variable terrain surfaces that change depending on weight applied, speed and etc.
8.) If Px4 does not have separate tunable parameters for "on the ground" and "in flight" control, where should I be modifying/adding in the firmware to do so.

Miscellaneous:
1.) Which flight modes do L1 damping/tuning affect and does it and/or can it scale differently on the ground vs in the air?
2.) Which modes do throttle slew rates affect and why would adjusting these throttle slew rates NOT change the actual throttle slew rate while in any mode. Are the throttle slew rates ignored in mission mode take-off roll? Are the throttle slew rates disabled on the ground but enabled in the air?

Mission Mode:
1.) If there is no take-off waypoint set, what checks are in place to prevent the aircraft from trying to climb up to the first waypoint? Can these checks be disabled?

dagar, 2017-06-09:
These are some good questions.

1. Currently only during auto TAKEOFF (runway only) and LANDING (after flair).
2. Only AUTO
3. Mostly on the ground, but there are brief sections where it's active in air. 
4. For runway takeoff the wheel steering is enabled until the climbout section above RWTO_NAV_ALT
5. Think of wheel steering as an alternate yaw controller, but because most hobby planes have steering and rudder on the same channel you have to switch between them.
6. Not very well.
7. No, but are you aware you can tune the wheel controller separately from yaw? FW_W_* parameters - https://dev.px4.io/en/advanced/parameter_reference.html#fw-attitude-control
8. If you think separate L1 parameters for ground vs air are necessary then you'd need change it in the fixed wing position controller. https://github.com/PX4/Firmware/blob/master/src/modules/fw_pos_control_l1/fw_pos_control_l1_main.cpp#L490

Misc
1. L1 is used in the position control manual mode and all AUTO modes.
2. 1 + altitude control mode. That parameter is used with the total energy control system (TECS) that's responsible for altitude and airspeed. The other modes are letting you pass stick input directly through to the throttle mixer. At that level you could use MOT_SLEW_MAX.

Mission mode
1. None

There are definitely improvements that could be made here, but I hope this clears up some of your questions for how it currently works.

tubeme, 2017-06-09:
For the Mission mode there are some checks BTW.

If your first Waypoint is further away MIS_DIST_1WP distance from Home it will not activate the mission. It is up to 10km. so no further than 10km.

stoneking, 2017-06-09:
Related question:  I tried switching from stabilized mode into mission mode on the ground during the takeoff role and observed an interruption of control and thrust for about a full second before mission mode kicked in.  Should I expect the same magnitude of interruption of control and thrust in the air, after takeoff?

dagar, 2017-06-09:
@stoneking no definitely not, and if you can share a log we can look at what actually happened. It may have just been the land detector.

stoneking, 2017-06-09:
Thanks.  I can't share that particular log, but if I repeat the experiment, I will try to save the log to share.

dagar, 2017-06-09:
No problem, just pay attention to the land detector. It produces the "Takeoff detected" and "Landing detected" messages in the console and can also be seem in the EXTENDED_SYS_STATE (http://mavlink.org/messages/common#EXTENDED_SYS_STATE) mavlink message.

tubeme, 2017-06-09:
@stoneking I can confirm that there is no such a stop in control during flight between all modes and mission. Mission just kicks in... Although Take Off Mode (Take off  and land are modes not roles in terms of PX4.) There might be some hick up we haven't tested. But without logs there is no solution to the problem.

Another thing to mention is that we did not have any problems with with holding our heading during take off on a runway. We did Stabalized mode take off and Position and Altitude  modes take off which went very smooth. Given we used the default values for the front wheel pids.

Regarding N6 question. There are two ways the planes control the heading on the runway with front gear if on the ground and with the rudder. So Given it can take off well in guided modes under pilot control should mean autonomous modes should be able to control the plane well. Even better than the pilot. We are preparing a plane for this experiment as well. We'll keep in touch.

closed 2018-02-21 by PX4BuildBot