PX4/PX4-Autopilot #7387: Issue with px4_add_module()

devbharat, 2017-06-09:
Doesn't `target_include_directories()` require `INTERFACE|PUBLIC|PRIVATE` as an argument? The definition of px4_add_module [here](https://github.com/PX4/Firmware/blob/5ee79648b7e709c0c31e4369f3f77ae39f1c7567/cmake/common/px4_base.cmake#L248) 

```
if(INCLUDES)
	target_include_directories(${MODULE} ${INCLUDES})
endif()
```

It exits with "target_include_directories called with invalid arguments"

dagar, 2017-06-09:
Do we actually need this functionality or would it make more sense to directly call include_directories() in that CMakeLists.txt?

devbharat, 2017-06-10:
Of course we could directly call target_include_directories() in the specific CMakeLists file, but I thought there was a point to all these px4 abstractions.

dagar, 2017-06-28:
In this case it really is just a wrapper and I'd rather strip away the additional complexity.

dagar, 2017-06-28:
That being said, I did the quick fix recently. https://github.com/PX4/Firmware/blob/master/cmake/common/px4_base.cmake#L248

closed 2017-06-28 by dagar