PX4/PX4-Autopilot #7390: 'fw_pos_control_l1 stop' crashes the system in v1.6.2

vvranjek, 2017-06-09:
Replicate the issue:

Flash v1.6.2 using qGroundControl-3.1.3
Open console and run:
`fw_pos_control_l1 stop`

Boot+crash:
```
[boot] Fault Log info File No 3 Length 3257 flags:0x01 state:1

[boot] Fault Log is Armed

sercon: Registering CDC/ACM serial driver
sercon: Successfully registered the CDC/ACM serial driver
[i] microSD mounted: /fs/microsd 
[hardfault_log] Fault Log is Armed

INFO  [param] selected parameter default file /fs/mtd_params
rgbled on I2C bus 2 at 0x55 (bus: 100 KHz, max: 100 KHz)
INFO  [load_mon] stack check enabled
ERROR [param] Parameter UAVCAN_ENABLE not found
INFO  [pwm_out_sim] MODE_16PWM
INFO  [mavlink] mode: Normal, data rate: 1200 B/s on /dev/ttyS1 @ 57600B
ERROR [param] Parameter UAVCAN_ENABLE not found
INFO  [ver] match: PX4FMU_V2
px4flow [169:100]
WARN  [px4flow] scanning I2C buses for device..
INFO  [mavlink] mode: Config, data rate: 800000 B/s on /dev/ttyACM0 @ 57600B
INFO  [logger] logger started (mode=all)
INFO  [init] Mixer: /etc/mixers/AERT.main.mix on /dev/pwm_output0 
INFO  [init] Mixer: /etc/mixers/pass.aux.mix appended to /dev/pwm_output0 

NuttShell (NSH)
nsh> [Kfw_pos_control_l1 stop
WARN  [fw_pos_control_l1] exiting.
up_hardfault: PANIC!!! Hard fault: 40000000
up_assert: Assertion failed at file:armv7-m/up_hardfault.c line: 171 task: fw_pos_control_l1
up_dumpstate: sp:     20005cc4
up_dumpstate: IRQ stack:
up_dumpstate:   base: 20005cf0
up_dumpstate:   size: 000002ec
up_dumpstate:   used: 00000134
up_stackdump: 20005cc0: 08098e8b 00000003 00000000 08098e93 08098e71 080a1d59 000000f0 2000b30c
up_stackdump: 20005ce0: 2001ab60 00000030 00000000 08098df7 00000000 10000010 00000001 00000000
up_dumpstate: sp:     2000b3e0
up_dumpstate: User stack:
up_dumpstate:   base: 2000b410
up_dumpstate:   size: 00000484
up_dumpstate:   used: 00000188
up_stackdump: 2000b3e0: 200001c8 0805800f 00000000 2001ab60 20003438 08058097 00000101 00000000
up_stackdump: 2000b400: 00000000 08092a89 00000000 00000000 deadbeef 2000b420 2000b432 00000000
up_registerdump: R0: 20015b20 2001b7b8 00000000 00000000 2001b080 2001ab60 00000030 00000000
up_registerdump: R8: 00000000 00000000 00000000 00000000 000000f0 2000b3e0 0808c10b 00000000
up_registerdump: xPSR: 60000000 BASEPRI: 000000f0 CONTROL: 00000000
up_registerdump: EXC_RETURN: ffffffe9
up_taskdump: Idle Task: PID=0 Stack Used=0 of 0
up_taskdump: hpwork: PID=1 Stack Used=848 of 1780
up_taskdump: lpwork: PID=2 Stack Used=640 of 1780
up_taskdump: init: PID=3 Stack Used=1720 of 2484
up_taskdump: log_writer_file: PID=196 Stack Used=344 of 1060
up_taskdump: dataman: PID=102 Stack Used=704 of 1180
up_taskdump: navigator: PID=231 Stack Used=856 of 1772
up_taskdump: sensors: PID=104 Stack Used=1304 of 1980
up_taskdump: commander: PID=106 Stack Used=2320 of 3644
up_taskdump: commander_low_prio: PID=107 Stack Used=592 of 2996
up_taskdump: mavlink_if1: PID=176 Stack Used=1600 of 2388
up_taskdump: mavlink_rcv_if1: PID=177 Stack Used=1576 of 2140
up_taskdump: pwm_out_sim: PID=114 Stack Used=760 of 1172
up_taskdump: px4io: PID=116 Stack Used=1016 of 1380
up_taskdump: ekf2: PID=215 Stack Used=1824 of 5780
up_taskdump: fw_pos_control_l1: PID=248 Stack Used=392 of 1156
up_taskdump: fw_att_control: PID=217 Stack Used=712 of 1476
up_taskdump: mavlink_if0: PID=124 Stack Used=1568 of 2436
up_taskdump: mavlink_rcv_if0: PID=125 Stack Used=1316 of 2140
up_taskdump: logger: PID=190 Stack Used=1272 of 3532
[boot] Fault Log info File No 3 Length 3257 flags:0x01 state:0
```

dagar, 2017-06-09:
Thanks for reporting, this isn't new. I was planning to fix it by moving the FW controllers over to the new common module base class once available. https://github.com/PX4/Firmware/pull/7145

dagar, 2018-02-17:
Please try https://github.com/PX4/Firmware/pull/8905.

closed 2018-02-17 by LorenzMeier