PX4/PX4-Autopilot #7392: GPS_STATUS message support

dswk, 2017-06-09:
PX4 does not appear to support the GPS_STATUS message. I was hoping to use MAVROS to gather the SNR (C/N0) values to each satellite in view from my multirotor in flight. Currently MAVROS itself does not support this message, which I'm hoping will change, but it can't add support until it is produced at the source (PX4). This is the type of info available in the $GPGSV NMEA 0183 message, which is the way I got access to this info in the past (not using a PX4), and at present if I want this info I would need to carry a second GPS receiver.

My GPS receiver is a uBlox Neo-M8N. The UBX-RXM-RAWX message does contain the C/N0 info for each satellite, among many other things, provided the receiver has protocol version 17 (in the docs I also see a note about this applying to "Time Sync products only").

I understand this isn't used by a lot of people, although there are some very interesting things that can be done when you know the SNR to each satellite in the sky. I would think people interested in RTK would want this same information, as that message contains the pseudoranges as well. Here are a couple of examples of published work using this type of information:

http://www.cc.gatech.edu/~summetj/papers/kim_iswc_08.pdf
https://tia.ucsb.edu/about-tia/success-stories/shadowmaps/
ftp://www.ngs.noaa.gov/pub/abilich/LarsonSnow_paperOverview.ppt

Here is the thread on the mavros Github repository where I started down this path:

https://github.com/mavlink/mavros/issues/717#issuecomment-307493951

Thanks!

TSC21, 2017-06-19:
@dswk maybe you will want to consider `GPS_RTK` msg instead?

dswk, 2017-06-19:
If this is what you’re referring to:

https://pixhawk.ethz.ch/mavlink/#GPS_RTK

It doesn’t appear to list the individual SNR, azimuth, and elevation for the individual satellites, which is what I need.

closed 2017-07-26 by TSC21