PX4/PX4-Autopilot #7393: ESC calibration always time out after connected to battery

stlfatboy, 2017-06-11:


sanderux, 2017-06-11:
What esc's are you using, how lang after boot should the signal go low?

stlfatboy, 2017-06-11:
@sanderux 
I am using blhell_s.
Not sure for that, I am new to drone.

sanderux, 2017-06-12:
@LorenzMeier @dagar 
Currently esc calibration is hardcoded to wait 3 secs after battery connected before setting the PWM signal low. (https://github.com/PX4/Firmware/blob/master/src/modules/commander/esc_calibration.cpp#L114)

According to the docs BLHeli needs at minimum 3 seconds of PWM high, and SimonK needs at maximum 2 seconds. in both cases we must account for the time it takes the ESC to boot.

Any ideas how to approach this?

mhkabir, 2017-06-12:
Related : DJI E6xx ESCs (and other DJI ones) run a self-check and pulse motors which takes 3-4 seconds on startup. This means that the calibration needs at-least 7-8 seconds of high signal after plugging in the battery.

I think a good way to expose the "high time" parameter to the user would be a dropdown in the QGC Power menu where the calibrate button is.

Related issues : 
https://github.com/PX4/Firmware/issues/6941
https://github.com/PX4/Firmware/issues/6943

sanderux, 2017-06-12:
Should this be a param or should it be part of http://mavlink.org/messages/common#MAV_CMD_PREFLIGHT_CALIBRATION and have the user select the timeout from QGC every time?

mhkabir, 2017-06-12:
I don't see a clean way to put it into the command, so a parameter would probably be the easiest thing to do.

stlfatboy, 2017-06-14:
So, no temporary solution until that time can be set by user? Is this possible if connected to batt before click the calibrate button?

mhkabir, 2017-06-14:
Increase this as a temporary fix : https://github.com/PX4/Firmware/blob/master/src/modules/commander/esc_calibration.cpp#L114

sanderux, 2017-06-14:
@mhkabir that will break it for most users. Simonk is far more common on a mc

mhkabir, 2017-06-14:
What I meant was that @stlfatboy could increase it for his setup just to get it working till we properly patch it master.

sanderux, 2017-06-14:
Right, yes that would work.
Seems like an easy patch to add a param, i can do it when i get back from holiday or you could take it on if you want to

sanderux, 2017-07-04:
@mhkabir @dagar ESC calibration is performed here:
https://github.com/PX4/Firmware/blob/master/src/modules/commander/esc_calibration.cpp#L77

Thats normally (through GCS) called from here:
https://github.com/PX4/Firmware/blob/13bf3122635a93df2e0d7036c292c173aafe8ce1/src/modules/commander/commander.cpp#L4248

But can also be called through a startup command from here:
https://github.com/PX4/Firmware/blob/13bf3122635a93df2e0d7036c292c173aafe8ce1/src/modules/commander/commander.cpp#L434

The calib script itself has no param loading so maybe best keep that. but from commander i could easily pass it for the normal calibration but the commandline calibration launches before params are loaded.

Any suggestions?

closed 2018-02-21 by PX4BuildBot