PX4/PX4-Autopilot #7394: 1.6.1 : No transition on SD CARD missing.

tubeme, 2017-06-11:
I dont know if it is VTOL problem but we tried it with VTOL. On SD CARD missing there was no sound or indication for the missing card. Also we were able to arm and fly the VTOL in MC mode without any problems. But then it did not allow us to transition in FW flight without the SD CARD. Also there was no indication in QGC about the missing card, nor there was a sound form the Pixhawk.

dagar, 2017-06-13:
This is one of those things that likely hasn't been tested. 

I've wasted time flying without an sdcard when I needed the data. Maybe it could be an optional parameter for arming?

tubeme, 2017-06-13:
This is one of those things that you find trying to find why there is no transition on a regular flight... changed an esc then again no transition until realizing somebody forgot to put in the sc-card :)

tubeme, 2017-06-13:
I think there should be NO flight without a SD CARD period!! In the days where drones are able to do so much damage there should always be a way to see what caused the damage...

dagar, 2017-06-13:
1. Transition not working with an sdcard sounds like a bug
2. You also need to make it work for systems like the Intel Aero that don't have sdcards.

tubeme, 2017-06-13:
Hmmm didn't they have some kind of internal memory? Other than the boards that don't support memory of some kind all other should fly with the SD CARD.  And before we've had a very disturbing sound from PX4 on card missing. It was with a RED light and could not arm. That is the perfect way it should be. If we had this alarm (as we are used to) we couldn't have been able to arm in the first place so no problem with the transition without sd-card.

The SD card is used to store the mission. If there is no card where do we store the mission?

closed 2018-02-21 by PX4BuildBot