PX4/PX4-Autopilot #7398: Mag STALE fail with external uavcan mag

tstellanova, 2017-06-12:
- Working with PX4 firmware tagged v1.6.2, with an external UAVCAN magnetometer,  during startup sequence, sensors reports:

```
ERROR [sensors] Mag #3 fail:  STALE!
```
From nsh command line, `sensors status` reports:

```
INFO  [sensors] mag status:
INFO  [lib__ecl] validator: best: 0, prev best: 3, failsafe: YES (1 events)
INFO  [lib__ecl] sensor #0, prio: 50, state: OK
INFO  [lib__ecl] 	val:   0.1031, lp:   0.1158 mean dev:  -0.0000 RMS:   0.0096 conf:   1.0000
INFO  [lib__ecl] 	val:   0.1886, lp:   0.1875 mean dev:   0.0000 RMS:   0.0083 conf:   1.0000
INFO  [lib__ecl] 	val:   0.4071, lp:   0.4194 mean dev:  -0.0000 RMS:   0.0075 conf:   1.0000
INFO  [lib__ecl] sensor #1, prio: 50, state: OK
INFO  [lib__ecl] 	val:   0.4108, lp:   0.4062 mean dev:   0.0024 RMS:   0.0224 conf:   1.0000
INFO  [lib__ecl] 	val:   0.1509, lp:   0.1436 mean dev:  -0.0021 RMS:   0.0200 conf:   1.0000
INFO  [lib__ecl] 	val:   0.5628, lp:   0.5659 mean dev:  -0.0009 RMS:   0.0155 conf:   1.0000
INFO  [lib__ecl] sensor #2, prio: 50, state: OK
INFO  [lib__ecl] 	val:   0.1090, lp:   0.1093 mean dev:  -0.0017 RMS:   0.0162 conf:   1.0000
INFO  [lib__ecl] 	val:   0.2009, lp:   0.2003 mean dev:   0.0004 RMS:   0.0041 conf:   1.0000
INFO  [lib__ecl] 	val:   0.4440, lp:   0.4446 mean dev:  -0.0007 RMS:   0.0074 conf:   1.0000
INFO  [lib__ecl] sensor #3, prio: 125, state: STALE
INFO  [lib__ecl] 	val:   0.0000, lp:   0.0000 mean dev:  -0.0003 RMS:   0.0069 conf:   0.0000
INFO  [lib__ecl] 	val:   0.0000, lp:   0.0000 mean dev:  -0.0027 RMS:   0.0166 conf:   0.0000
INFO  [lib__ecl] 	val:   0.0000, lp:   0.0000 mean dev:  -0.0056 RMS:   0.0352 conf:   0.0000
```

- Prior to v1.6.2 I did not see this error.
- Verified with two different Zubax GNSS2 sensors connected via UAVCAN.  Both previously worked fine with eg v.1.6.0

@mhkabir Perhaps you have some ideas.

mhkabir, 2017-06-12:
Related #7397, but I'm not sure why it happens yet.

mhkabir, 2017-06-12:
OK I think the rotation or something goes wrong because all the values are zero.

tstellanova, 2017-06-12:
More info:

- Build for pixhawk 2.1 hardware (px4fmu-v3_default)
- Tried recalibrating mag.  Calibration succeeds, but `sensor status` still shows STALE and zeros for `val` and `lp`

mhkabir, 2017-06-12:
Hmm, I tried on v4 hardware (Pixracer) with Zubax GNSS 2 and I couldn't reproduce : 
```
INFO  [sensors] mag status:
INFO  [lib__ecl] validator: best: 2, prev best: 0, failsafe: YES (1 events)
INFO  [lib__ecl] sensor #0, prio: 100, state: OK
INFO  [lib__ecl] 	val:   0.3014, lp:   0.2978 mean dev:   0.0198 RMS:   0.1006 conf:   1.0000
INFO  [lib__ecl] 	val:  -0.2960, lp:  -0.2947 mean dev:   0.0112 RMS:   0.0583 conf:   1.0000
INFO  [lib__ecl] 	val:   0.1534, lp:   0.1604 mean dev:   0.0056 RMS:   0.0606 conf:   1.0000
INFO  [lib__ecl] sensor #1, prio: 50, state: OK
INFO  [lib__ecl] 	val:   0.2307, lp:   0.2448 mean dev:   0.0095 RMS:   0.0733 conf:   1.0000
INFO  [lib__ecl] 	val:  -0.2609, lp:  -0.2627 mean dev:  -0.0024 RMS:   0.0464 conf:   1.0000
INFO  [lib__ecl] 	val:   0.0143, lp:   0.0199 mean dev:  -0.0177 RMS:   0.0589 conf:   1.0000
INFO  [lib__ecl] sensor #2, prio: 125, state: OK
INFO  [lib__ecl] 	val:   0.3892, lp:   0.3846 mean dev:   0.0340 RMS:   0.1012 conf:   1.0000
INFO  [lib__ecl] 	val:  -0.2878, lp:  -0.2888 mean dev:  -0.0139 RMS:   0.0684 conf:   1.0000
INFO  [lib__ecl] 	val:   0.2061, lp:   0.2083 mean dev:  -0.0210 RMS:   0.0758 conf:   1.0000
```
Can you give me a param dump _or_ tell me the real rotations of the autopilot and GNSS unit + `CAL_MAG_ROT` params?

mhkabir, 2017-06-12:
Reproduced on Pixhawk 2.1 : 

```
INFO  [sensors] mag status:
INFO  [lib__ecl] validator: best: 0, prev best: 3, failsafe: YES (2 events)
INFO  [lib__ecl] sensor #0, prio: 50, state: OK
INFO  [lib__ecl] 	val:   0.4231, lp:   0.4379 mean dev:  -0.0018 RMS:   0.0144 conf:   1.0000
INFO  [lib__ecl] 	val:   0.1072, lp:   0.1106 mean dev:   0.0014 RMS:   0.0099 conf:   1.0000
INFO  [lib__ecl] 	val:   0.5650, lp:   0.5632 mean dev:   0.0003 RMS:   0.0106 conf:   1.0000
INFO  [lib__ecl] sensor #1, prio: 50, state: OK
INFO  [lib__ecl] 	val:   0.4226, lp:   0.4242 mean dev:  -0.0023 RMS:   0.0064 conf:   1.0000
INFO  [lib__ecl] 	val:  -0.0951, lp:  -0.0948 mean dev:  -0.0066 RMS:   0.0273 conf:   1.0000
INFO  [lib__ecl] 	val:   0.1834, lp:   0.1842 mean dev:  -0.0142 RMS:   0.0529 conf:   1.0000
INFO  [lib__ecl] sensor #2, prio: 50, state: OK
INFO  [lib__ecl] 	val:   0.1693, lp:   0.1502 mean dev:  -0.0117 RMS:   0.0363 conf:   1.0000
INFO  [lib__ecl] 	val:  -0.4654, lp:  -0.4681 mean dev:  -0.0129 RMS:   0.0550 conf:   1.0000
INFO  [lib__ecl] 	val:  -0.0665, lp:  -0.0720 mean dev:  -0.0095 RMS:   0.0413 conf:   1.0000
INFO  [lib__ecl] sensor #3, prio: 125, state: STALE
INFO  [lib__ecl] 	val:   0.0000, lp:   0.0000 mean dev:  -0.0126 RMS:   0.0356 conf:   0.0000
INFO  [lib__ecl] 	val:   0.0000, lp:  -0.0000 mean dev:   0.0333 RMS:   0.0880 conf:   0.0000
INFO  [lib__ecl] 	val:   0.0000, lp:  -0.0000 mean dev:   0.0081 RMS:   0.0234 conf:   0.0000
```

nicolaerosia, 2017-06-12:
The actual related issue is https://github.com/PX4/Firmware/issues/7395

tstellanova, 2017-06-12:
FWIW, this external mag has no rotation relative to the board (it has the same orientation as the Pixhawk 2.1)

closed 2017-06-20 by mhkabir