PX4/PX4-Autopilot #7399: Snapdragon QuRT compiler optimization

dagar, 2017-06-12:
Continuing from https://github.com/PX4/Firmware/pull/7359

I've been gradually trying to simplifying the PX4 build system as its increasing complexity results in hiding issues and things not working as intended.

One example is the optimization level for QuRT.

In cmake/qurt/px4_impl_qurt.cmake (https://github.com/PX4/Firmware/blob/stable/cmake/qurt/px4_impl_qurt.cmake#L184) the optimization level is forced -O0, but 'added_cflags' is the wrong variable name, and either way -O2 is prepended.

@jywilson @mcharleb do you want to do a pass and determine what should be set on QuRT based on toolchain version?

mcharleb, 2017-08-28:
I think the main issue was EKF takes a LONG time to compile for QuRT when -O2 is enabled. The second issue was that cmake _hexagon is for building more than just PX4 so a default level of -O0 was set in the toolchain file. I'd have to make sure removing it didn't have an adverse impact on other projects. @dagar Why should PX4 override the optimization level set by the toolchain.cmake file?

dagar, 2017-08-29:
The original issue here was more that it was being set in about 3 different places to different values.

Usually it's left to the project and set by CMAKE_BUILD_TYPE. Some IDEs will use both a Debug and Release configuration.
If you wanted to force no optimization you could have QuRT set the build type to debug somewhere, or you could remove the optimization from only EKF2. https://github.com/PX4/Firmware/blob/master/CMakeLists.txt#L156

closed 2018-02-21 by PX4BuildBot