PX4/PX4-Autopilot #7402: PX4 1.6.1: Land detector did not disarm after land in mission VTOL

tubeme, 2017-06-12:
Hello,

We made a mission without RC control only with the QGC. We used version 3.1.3. Everithing went ok except the land detector did not disarm after the mission was complete and the VTOL was landed. The only way we could disarm with 3.1.3 was emergency stop.

VTOL left on the ground with spinning motors for some time.

Here is the log:
http://review.px4.io/plot_app?log=971f681b-2e49-4e67-816f-5b3b9c52a817

Also in Position or Altitude modes we have the same problem. The land detector probably is working, because when we touch down in one of Alt and Pos Modes the QGC says we have to be either in manual mode or landed in order to disarm.

I saw it is very easy in QGC 3.2 to disarm. But it is still very nice if in the case of mission and Guided modes after touch down the land detector could at least disarm in 10-20 seconds.

sanderux, 2017-06-13:
@tubeme i seem to remember advising you to lower your params to avoid false positives on backtransitions. did you check if that is related?

tubeme, 2017-06-13:
@sanderux No it is not related. We don't have this false alarms on back transition since you gave us the advice. The problem is that it does not disarm once landed after mission or in Position/Altitude modes.

LorenzMeier, 2017-06-14:
The issue is most likely how idle throttle and other detection parameters are set up. It is very reliable if configured for the airframe.

mhkabir, 2017-06-14:
@LorenzMeier The setup/tuning of land detector for a particular airframe needs documentation. There's nothing on that side.

LorenzMeier, 2017-06-14:
https://dev.px4.io/en/tutorials/land_detector.html

mhkabir, 2017-06-14:
Ah, thanks. My bad.

LorenzMeier, 2017-06-14:
I've added a bit more detail and I hope that this is now enough for @tubeme to tune it. You can try increasing `LNDMC_THR_RANGE` slightly. Other than that we would need to test if RC is implicitly required to auto-disarm, since it is illegal in most jurisdictions to fly without one we haven't extensively tested this.

tubeme, 2017-06-14:
@LorenzMeier Thanks Lorenz, before I posted here I read very carefully the doc you sent me, but I did not find the answer there. Also we are aware of LNDMC_THR_RANGE parameter and it is OK with the current setting for us. We dont' expect to have the detector disarm when the thr is above this parameter. 

Today we will make a test of landing in Position and follow the whole procedure so to see if the land detector disarms with RC. Our previous test on this showed that the land detector does not disarm on Position landing and "0" thorttle. But we always accept that we did something wrong so we will make another test.

Speaking of legal matters regarding the control, there is no real definitions yet of what is legal or not in a formal way. This includes EU and US. We follow the topic very closely of what is going on in the EU and US govs. What is known for now is that it is illegal to have the aircraft without CONTROL. Legally speaking having the ground station connected to the aircraft gives you very much all the control you need legally speaking. And there are numerous examples of commercial products that are operated without RC control. Illegal is to have both the RC control switched off and ground station control switched off at the same time, ie remove the circuit breakers and remove the RTL on RC AND telemetry loss. We did this experiment as well but we do it in the dessert far far away from the range of the aircraft to a populated areas. So to speak, having a RTL setup only on the telemetry loss and operating the aircraft from the QGC is perfectly legal, and most likely will be defined in this way, because there are more and more crafts produced to have only ground station control. Personally speaking I think that giving RC control to a novice person is much more dangerous than giving them the option to fly with ground station only.

LorenzMeier, 2017-06-19:
@tubeme Do you have new logs or an update on this? I would like to help you getting this resolved quickly and fix the documentation where required. It should be easy to configure and if its not we need to fix it.

tubeme, 2017-06-22:
We were not able to reproduce anymore. It appears only in strong weather when the wind gusts are creating some lift in the wings on the ground, combined with the rotation of the quad props.

LorenzMeier, 2017-06-23:
Ok. There is nothing we can do in those conditions except using a Lidar for the detection. Is this something you would want to contribute to?

closed 2017-06-23 by LorenzMeier