PX4/PX4-Autopilot #7405: 1.6.2 VTOL RTL does not trigger back-transition

bartslinger, 2017-06-13:
I think that this commit breaks the back-transition when RTL is commanded to a VTOL drone:
https://github.com/PX4/Firmware/pull/7249

However I'm not 100% sure since my drone doesn't work with master or stable. @dagar suggested that this could easily be tested in SITL, although I don't have that setup at the moment. I would appreciate it if someone can verify this with SITL.

LorenzMeier, 2017-06-14:
@sanderux The comment you made on the PR is I think in the wrong location. What I removed there was clearly wrong and added only a few weeks before that. I think its much rather related to the other handling changes. If you still think its that code part please explain in more detail.

sanderux, 2017-06-14:
I'll try and reproduce in sitl. It could also be that no landing was set for rtl. @bartslinger do you have a log?

dagar, 2017-06-27:
This is resolved.

closed 2017-06-27 by dagar