PX4/PX4-Autopilot #7406: CPU Temperature reporting

nicolaerosia, 2017-06-13:
When playing with Navio2 on RPI3, I experienced a sudden reset which I believe is caused by high temperature.
Is there a message where we could inject the CPU Temperature so we can monitor it from GCS?

nicolaerosia, 2017-06-17:
@mhkabir @bkueng @LorenzMeier I need this before my next trial, any suggestions?

LorenzMeier, 2017-06-17:
Can you spec out a new packet for this? I'm happy to add it to the common spec. You could also take some inspiration from this:
https://github.com/pixhawk/mavconn/blob/master/src/core/mavconn-core.cc#L397-L449

mhkabir, 2017-06-17:
Hi @nicolaerosia, I've been working on a thermal control module for IMUs (e.g on the Pixhawk 2). Maybe it makes sense to centralise that work together and yours?

closed 2018-02-21 by PX4BuildBot