PX4/PX4-Autopilot #7409: Out-of-tree module build is broken

mhkabir, 2017-06-13:
I was trying out the out-of-tree module build functionality (https://dev.px4.io/en/advanced/out_of_tree_modules.html), but it seems to be broken at the moment. I suspect that this is due to the recent param parser refactoring.

```
Mohammeds-MacBook-Pro:Firmware kabir$ make px4fmu-v3_default EXTERNAL_MODULES_LOCATION=/Users/kabir/Development/PX4/uasys-px4-modules
-- CMAKE_MODULE_PATH: /Users/kabir/Development/PX4/Firmware/cmake
-- Nuttx build for px4fmu-v3 on m4 hardware, using nsh with ROMFS on px4fmu_common
-- Build Type: MinSizeRel
-- PX4 VERSION: v1.6.2-37-ge6159d9e5
-- CONFIG: nuttx-px4fmu-v3-default
CMake Deprecation Warning at /usr/local/Cellar/cmake/3.8.2/share/cmake/Modules/CMakeForceCompiler.cmake:69 (message):
  The CMAKE_FORCE_C_COMPILER macro is deprecated.  Instead just set
  CMAKE_C_COMPILER and allow CMake to identify the compiler.
Call Stack (most recent call first):
  cmake/toolchains/Toolchain-arm-none-eabi.cmake:31 (cmake_force_c_compiler)
  /usr/local/Cellar/cmake/3.8.2/share/cmake/Modules/CMakeDetermineSystem.cmake:88 (include)
  CMakeLists.txt:273 (project)

CMake Deprecation Warning at /usr/local/Cellar/cmake/3.8.2/share/cmake/Modules/CMakeForceCompiler.cmake:83 (message):
  The CMAKE_FORCE_CXX_COMPILER macro is deprecated.  Instead just set
  CMAKE_CXX_COMPILER and allow CMake to identify the compiler.
Call Stack (most recent call first):
  cmake/toolchains/Toolchain-arm-none-eabi.cmake:37 (cmake_force_cxx_compiler)
  /usr/local/Cellar/cmake/3.8.2/share/cmake/Modules/CMakeDetermineSystem.cmake:88 (include)
  CMakeLists.txt:273 (project)

CMake Deprecation Warning at /usr/local/Cellar/cmake/3.8.2/share/cmake/Modules/CMakeForceCompiler.cmake:69 (message):
  The CMAKE_FORCE_C_COMPILER macro is deprecated.  Instead just set
  CMAKE_C_COMPILER and allow CMake to identify the compiler.
Call Stack (most recent call first):
  cmake/toolchains/Toolchain-arm-none-eabi.cmake:31 (cmake_force_c_compiler)
  build_px4fmu-v3_default/CMakeFiles/3.8.2/CMakeSystem.cmake:6 (include)
  CMakeLists.txt:273 (project)

CMake Deprecation Warning at /usr/local/Cellar/cmake/3.8.2/share/cmake/Modules/CMakeForceCompiler.cmake:83 (message):
  The CMAKE_FORCE_CXX_COMPILER macro is deprecated.  Instead just set
  CMAKE_CXX_COMPILER and allow CMake to identify the compiler.
Call Stack (most recent call first):
  cmake/toolchains/Toolchain-arm-none-eabi.cmake:37 (cmake_force_cxx_compiler)
  build_px4fmu-v3_default/CMakeFiles/3.8.2/CMakeSystem.cmake:6 (include)
  CMakeLists.txt:273 (project)

-- The ASM compiler identification is GNU
-- Found assembler: /usr/local/bin/arm-none-eabi-gcc
-- Found PythonInterp: /usr/local/bin/python (found version "2.7.13") 
-- Found PY_jinja2: /usr/local/lib/python2.7/site-packages/jinja2  
-- C compiler: arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 5.4.1 20160919 (release) [ARM/embedded-5-branch revision 240496]
-- C++ compiler: arm-none-eabi-g++ (GNU Tools for ARM Embedded Processors) 5.4.1 20160919 (release) [ARM/embedded-5-branch revision 240496]
-- External modules: /Users/kabir/Development/PX4/uasys-px4-modules
-- Using C++03
-- Release build type: MinSizeRel
-- Adding UAVCAN STM32 platform driver
-- Adding ROMFS on px4fmu-v3
-- Nuttx build for px4fmu-v3 on m3 hardware, using nsh 
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/kabir/Development/PX4/Firmware/build_px4fmu-v3_default
[12/766] Generating ../../../../parameters.xml
FAILED: parameters.xml 
cd /Users/kabir/Development/PX4/Firmware/build_px4fmu-v3_default && /usr/local/bin/python /Users/kabir/Development/PX4/Firmware/Tools/px_process_params.py -s /Users/kabir/Development/PX4/Firmware/src/drivers/airspeed /Users/kabir/Development/PX4/Firmware/src/drivers/blinkm /Users/kabir/Development/PX4/Firmware/src/drivers/bmi160 /Users/kabir/Development/PX4/Firmware/src/drivers/bmp280 /Users/kabir/Development/PX4/Firmware/src/drivers/boards/px4fmu-v2 /Users/kabir/Development/PX4/Firmware/src/drivers/bst /Users/kabir/Development/PX4/Firmware/src/drivers/camera_trigger /Users/kabir/Development/PX4/Firmware/src/drivers/device /Users/kabir/Development/PX4/Firmware/src/drivers/ets_airspeed /Users/kabir/Development/PX4/Firmware/src/drivers/frsky_telemetry /Users/kabir/Development/PX4/Firmware/src/drivers/gps /Users/kabir/Development/PX4/Firmware/src/drivers/hmc5883 /Users/kabir/Development/PX4/Firmware/src/drivers/hott /Users/kabir/Development/PX4/Firmware/src/drivers/hott/hott_sensors /Users/kabir/Development/PX4/Firmware/src/drivers/hott/hott_telemetry /Users/kabir/Development/PX4/Firmware/src/drivers/iridiumsbd /Users/kabir/Development/PX4/Firmware/src/drivers/l3gd20 /Users/kabir/Development/PX4/Firmware/src/drivers/led /Users/kabir/Development/PX4/Firmware/src/drivers/lis3mdl /Users/kabir/Development/PX4/Firmware/src/drivers/ll40ls /Users/kabir/Development/PX4/Firmware/src/drivers/lsm303d /Users/kabir/Development/PX4/Firmware/src/drivers/mb12xx /Users/kabir/Development/PX4/Firmware/src/drivers/meas_airspeed /Users/kabir/Development/PX4/Firmware/src/drivers/mkblctrl /Users/kabir/Development/PX4/Firmware/src/drivers/mpu6000 /Users/kabir/Development/PX4/Firmware/src/drivers/mpu9250 /Users/kabir/Development/PX4/Firmware/src/drivers/ms5611 /Users/kabir/Development/PX4/Firmware/src/drivers/oreoled /Users/kabir/Development/PX4/Firmware/src/drivers/pwm_input /Users/kabir/Development/PX4/Firmware/src/drivers/pwm_out_sim /Users/kabir/Development/PX4/Firmware/src/drivers/px4flow /Users/kabir/Development/PX4/Firmware/src/drivers/px4fmu /Users/kabir/Development/PX4/Firmware/src/drivers/px4io /Users/kabir/Development/PX4/Firmware/src/drivers/rgbled /Users/kabir/Development/PX4/Firmware/src/drivers/sf0x /Users/kabir/Development/PX4/Firmware/src/drivers/sf1xx /Users/kabir/Development/PX4/Firmware/src/drivers/snapdragon_rc_pwm /Users/kabir/Development/PX4/Firmware/src/drivers/srf02 /Users/kabir/Development/PX4/Firmware/src/drivers/stm32 /Users/kabir/Development/PX4/Firmware/src/drivers/stm32/adc /Users/kabir/Development/PX4/Firmware/src/drivers/stm32/tone_alarm /Users/kabir/Development/PX4/Firmware/src/drivers/tap_esc /Users/kabir/Development/PX4/Firmware/src/drivers/trone /Users/kabir/Development/PX4/Firmware/src/drivers/vmount /Users/kabir/Development/PX4/Firmware/src/modules/sensors /Users/kabir/Development/PX4/Firmware/src/systemcmds/bl_update /Users/kabir/Development/PX4/Firmware/src/systemcmds/config /Users/kabir/Development/PX4/Firmware/src/systemcmds/dumpfile /Users/kabir/Development/PX4/Firmware/src/systemcmds/esc_calib /Users/kabir/Development/PX4/Firmware/src/systemcmds/hardfault_log /Users/kabir/Development/PX4/Firmware/src/systemcmds/led_control /Users/kabir/Development/PX4/Firmware/src/systemcmds/mixer /Users/kabir/Development/PX4/Firmware/src/systemcmds/motor_ramp /Users/kabir/Development/PX4/Firmware/src/systemcmds/mtd /Users/kabir/Development/PX4/Firmware/src/systemcmds/nshterm /Users/kabir/Development/PX4/Firmware/src/systemcmds/param /Users/kabir/Development/PX4/Firmware/src/systemcmds/perf /Users/kabir/Development/PX4/Firmware/src/systemcmds/pwm /Users/kabir/Development/PX4/Firmware/src/systemcmds/reboot /Users/kabir/Development/PX4/Firmware/src/systemcmds/sd_bench /Users/kabir/Development/PX4/Firmware/src/systemcmds/top /Users/kabir/Development/PX4/Firmware/src/systemcmds/topic_listener /Users/kabir/Development/PX4/Firmware/src/systemcmds/ver /Users/kabir/Development/PX4/Firmware/src/drivers/sf0x/sf0x_tests /Users/kabir/Development/PX4/Firmware/src/drivers/test_ppm /Users/kabir/Development/PX4/Firmware/src/modules/commander/commander_tests /Users/kabir/Development/PX4/Firmware/src/lib/controllib/controllib_test /Users/kabir/Development/PX4/Firmware/src/modules/mavlink/mavlink_tests /Users/kabir/Development/PX4/Firmware/src/modules/mc_pos_control/mc_pos_control_tests /Users/kabir/Development/PX4/Firmware/src/modules/unit_test /Users/kabir/Development/PX4/Firmware/src/modules/uORB/uORB_tests /Users/kabir/Development/PX4/Firmware/src/systemcmds/tests /Users/kabir/Development/PX4/Firmware/src/modules/commander /Users/kabir/Development/PX4/Firmware/src/modules/events /Users/kabir/Development/PX4/Firmware/src/modules/gpio_led /Users/kabir/Development/PX4/Firmware/src/modules/land_detector /Users/kabir/Development/PX4/Firmware/src/modules/load_mon /Users/kabir/Development/PX4/Firmware/src/modules/mavlink /Users/kabir/Development/PX4/Firmware/src/modules/navigator /Users/kabir/Development/PX4/Firmware/src/modules/uavcan /Us [... cut ...]
    if not scanner.ScanDir(args.src_path, parser):
  File "/Users/kabir/Development/PX4/Firmware/Tools/px4params/srcscanner.py", line 20, in ScanDir
    for filename in os.listdir(srcdir):
OSError: [Errno 2] No such file or directory: '/Users/kabir/Development/PX4/Firmware/src//Users/kabir/Development/PX4/uasys-px4-modules/src/drivers/camera_trigger'
[14/766] Copying NuttX for px4fmu-v3 with nsh
ninja: build stopped: subcommand failed.
make: *** [px4fmu-v3_default] Error 1
Mohammeds-MacBook-Pro:Firmware kabir$ 

```

dagar, 2017-06-13:
Sorry, should be quick to fix though.
Can you roughly list the layout of the out of tree portion?

mhkabir, 2017-06-13:
Yeah, so the root directory is `/Users/kabir/Development/PX4/uasys-px4-modules`. It contains a directory `src`, which contains a `CMakeLists.txt` (see below) and a `drivers` directory.
I'm just trying to build the unmodified camera trigger driver from Firmware. It is located at `/Users/kabir/Development/PX4/uasys-px4-modules/src/drivers/camera_trigger`.

`CMakeLists.txt` : 

```
set(config_module_list_external
    drivers/camera_trigger
    PARENT_SCOPE
    )
```

LorenzMeier, 2017-10-05:
Fixed by Beat.

closed 2017-10-05 by LorenzMeier