PX4/PX4-Autopilot #7412: Bottle_drop not generating any AUX output

myeabhishek, 2017-06-14:
Hi Guys,
I am using bottle_drop module in PX4 to drop a payload from a moving Multicopter. I am using "bottle_drop drop" command from NSH shell to drop the payload. The problem is that I am not receiving any PWM output on any of the AUX pins required for my release mechanism on issuing this command. 

I have enabled and started this module for multicopters as I understand it was intended for fixed wings. Even then it should have no problem in generating a PWM output at the required time when this command is given. 

Please help!

LorenzMeier, 2017-06-14:
You need to create a mixer that connects the payload actuator control group with the PWM outputs.

LorenzMeier, 2017-06-14:
I'm closing this. Please post if you don't find in http://dev.px4.io how to add that mixer.

closed 2017-06-14 by LorenzMeier