PX4/PX4-Autopilot #7421: System time is wrong even when GPS time is valid

tstellanova, 2017-06-14:
Testing with PX4 firmware tagged v1.6.2 on Pixhawk 2.1 with working GPS hardware providing valid time, eg:

```
nsh> listener vehicle_gps_position 

TOPIC: vehicle_gps_position instance 0 #1
timestamp: 26954757
lat: 377600989
lon: -1224114390
alt: 31203
alt_ellipsoid: 1436
s_variance_m_s:   2.1113
c_variance_rad:   4.8381
fix_type: 3
eph:   8.5513
epv:  10.9430
hdop:   3.1699
vdop:   3.1699
noise_per_ms: 0
jamming_indicator: 0
vel_m_s:   0.8601
vel_n_m_s:  -0.1010
vel_e_m_s:  -0.6528
vel_d_m_s:  -0.5508
cog_rad:  -1.7243
vel_ned_valid: True
time_utc_usec: 1497483294299637
satellites_used: 7
```

The system time is not being set correctly from the GPS time:

```
nsh> date
Sat, Jan 01 01:31:05 2000
```
This results in eg `logger` log file timestamps being incorrect.

tstellanova, 2017-06-14:
More info:  Tested with UAVCAN GPS (Zubax GNSS 2.0)

tstellanova, 2017-06-15:
It appears that the gps driver calls `px4_clock_settime` but the uavcan driver does not.  It might be more appropriate for a listener to the `vehicle_gps_position` uORB topic to set the time? Not sure.

mhkabir, 2017-06-15:
Please just add the `clock_settime()` call to the UAVCAN GPS bridge. Awaiting the pull request :)

tstellanova, 2017-06-15:
PR submitted. FYI, from what I can see, the existing GPS drivers are not one-shot: They set the system clock over and over, on every GPS packet-- see the calls to `mktime` in ubx.cpp, mtk.cpp, and ashtech.cpp

Going forward, it might be better to have a single module (Commander?) that subscribes to `vehicle_gps_position` and calls `px4_clock_settime`, rather than requiring duplicate code in every GPS driver. This would mean that we'd be limited to setting the system clock with microsecond accuracy (from the `time_utc_usec` field) but would remove the need to call `mktime` in multiple drivers. 

However, there may be some complications of this I'm not aware of -- for instance, if you don't want `vehicle_gps_position` messages to dictate the system clock on HIL/SITL setups.

PR #7427 (2017-06-16): Set system real time clock once from GNSS data.
Fixes #7421

Tested with Pixracer and Zubax GNSS2.0

@mhkabir 
@LorenzMeier
commit fd47e0cb (2017-06-16): Set system real time clock once from GNSS data.

Fixes #7421

Tested with Pixracer and Zubax GNSS2.0
closed 2017-06-16 by mhkabir