PX4/PX4-Autopilot #7426: MC Autoland tries to go back home while descending

nkhoit, 2017-06-15:
Log: http://logs.px4.io/plot_app?log=53d0a6de-838a-4c26-b53f-87b6f19b23e5
I'm on master at bc406a1

Seems that autoland tries to control to an XY local position setpoint of (0,0).

Stifael, 2017-06-19:
@nkhoit, I think there are two questions which I also would like to have answered.
1.) is it necessary to control xy during auto land
advantage:
- the vehicle will not drift
disadvantage:
- the vehicle can only land if it has at least local position estimate in xy.
2.) why does it fly to home_position for  land
- I don't understand why it ever should do that for auto land. 
 a.) home_pos requires gps -> but auto land could also be done with flow, which uses local coordinate
 b.) if the decision is to allow auto land only for gps, then at least it could use the the current global position estimate instead of home_pos

In fact, the above questions also apply for takeoff and so on. auto takeoff should be allowed even if there is no global position estimate available. 

Some of the questions above have been addressed [here](https://github.com/PX4/Firmware/pull/7432), where the goal is to work with local coordinates within the navigator.

ChristophTobler, 2017-06-19:
I am able to reproduce this in SITL by starting in manual/position and then `commander land` (or using QGC). However it doesn't happen when in auto mode (e.g. hold). Will look into it

ChristophTobler, 2017-06-19:
fix: https://github.com/PX4/Firmware/pull/7441

closed 2017-06-19 by LorenzMeier