PX4/PX4-Autopilot #7431: Issues found with Pixhawk 3 Pro

bkueng, 2017-06-16:
While building a Quad (with @MaEtUgR & @ChristophTobler ) we encountered the following issues with Pixhawk 3 Pro on current master:

1) Oneshot does not work (by setting PWM_RATE to 0), using Hobbywing XRotor 35A Micro 3-6S BLHeli ESC's. Symptoms: when changing the rate to 0 and rebooting, there is no pwm output anymore. Needs further investigations. Is this supposed to work with an IO? @davids5 @kd0aij 
2) The FMU DEBUG port outputs garbage values when only the battery power source is connected. By attaching USB it works (using the same baudrate). Is this a GND problem?
3) motors turn when rebooting (in disarmed state). This only happened once, and we did not try to reproduce. @davids5 does the board have the same problem as pixracer (https://github.com/PX4/Firmware/issues/7123)?
4) The board has 2 power connectors. If power is connected to `Power 1`, QGC correctly shows the battery level, but the board cannot be armed (An error pops up that no power is connected), I traced it to [this condition](https://github.com/PX4/Firmware/blob/master/src/modules/commander/commander.cpp#L2036), `system_power` is published from [adc.cpp](https://github.com/PX4/Firmware/blob/master/src/drivers/stm32/adc/adc.cpp#L381). When connecting to `Power 2`, arming works, but QGC does not show the battery level. @davids5 can you look into it please? Preferably everything should work when connecting to Power 1.

FYI @jperin @klopezal

klopezal, 2017-06-16:
Thanks for reporting! A few comments : 

2. What pins are you using on the DEBUG port? It should be TX, RX and GND. GND must be common between all your devices : your USB-to-Serial converter needs the same reference in order to measure correct voltage differences. 

4. The architecture on FMUv4pro has changed, powering from servo rail is not possible anymore (without an external wiring). Two power bricks are available and a dedicated IC constantly checks for valid power supplies. They are prioritized in the following order : Brick 1 -> Brick 2 -> USB. Commander should use [this](https://github.com/PX4/Firmware/blob/master/src/drivers/boards/px4fmu-v4pro/board_config.h#L321) instead of servo rail check and change its logic : we should allow flying with Brick 1, Brick 2 and Brick 1+2, right?

I think that ideally we should be capable of telling the user which port is being used (imagine you are flying with two power modules and one starts failing, this could be unnoticed but potentially really bad). We should also manage the case where the user is using two different batteries and display voltage of the battery currently in use (or both). What do you think?

davids5, 2017-06-16:
@bkueng,

 I need a few more details.

1. What version HW?
2. RE:The FMU DEBUG port outputs garbage 
       What do you have connected to the debug port? Dronecode, or just FTDI cable?
3. RE:Motors turn when...
   What all was connected and on what pins.
4. What was the SYS_AUTOSTART value

I think the 2 (n) Brick management needs to be added to the code base.

bkueng, 2017-06-16:
> What pins are you using on the DEBUG port? It should be TX, RX and GND. GND must be common between all your devices : your USB-to-Serial converter needs the same reference in order to measure correct voltage differences.

Yes I use these. The thing is that it works if I connect USB power. Is the GND from the DEBUG port not connected to GND of the battery source?

Power detection: I agree we need to support both (n) bricks, with correct failover behavior. But for now I'm just after a quick solution so that at least one of the ports can be used consistently.

@davids5
It's V4PRO, the setup is documented here: https://dev.px4.io/en/airframes_multicopter/dji_flamewheel_450.html, SYS_AUTOSTART=4011.
I'm using a standard FTDI cable.

davids5, 2017-06-16:
@bkueng HW wize there is V1.0, V1.2, V1.2 with HW mods, V1.3. When was the board received?

LorenzMeier, 2017-06-16:
V1.3

davids5, 2017-06-16:
@klopezal Was a unit and cables shipped to me?

klopezal, 2017-06-16:
Yes!

bkueng, 2017-06-19:
A solution for 4., until we have support for N bricks, is here: #7442

davids5, 2017-06-30:
N Bricks is now supported in #7219.

closed 2017-08-20 by LorenzMeier