PX4/PX4-Autopilot #7439: v1.6.2  ignore RTL

svpcom, 2017-06-18:
I've have a two flights (one after one)
First flight was ok (mission and RTL): http://logs.px4.io/plot_app?log=f0d1fdb4-3c11-4f2a-8dc2-c0ba21947fd2
But in the second flight px4 ignore my RTL command: 
http://logs.px4.io/plot_app?log=c85d052b-70e7-4a59-98ba-6100d0934abf

LorenzMeier, 2017-06-18:
On a first pass on the log: in the first flight the RC switch went to 0.4 for RTL, but in the second it went to 0.6 and stayed in mission (where loitering is the default). If that is not consistent with your flight experience I can have another pass, but I suspect that either the operator didn't hit the same switch position or the RC switch thresholds sit right on the channel boundary. At any rate the commanded RTL was not the same RC switch position between the flights.

[image]

[image]

svpcom, 2017-06-18:
I have a single channel for flight mode (manual/stab/mission) and RTL.
```
RC_ASSIST_TH	0.147499993443489075
RC_POSCTL_TH	0.250000000000000000
RC_AUTO_TH	0.332500010728836060
RC_LOITER_TH	0.414999991655349731
RC_RETURN_TH	0.500000000000000000
```
My radio sum state of switches with different coefficients so actual value of RTL will depends from other switches but always > 0.5

Also selected region colored in plot as "RTL mode" and there are a log entries:
```
0:18:19 WARNING [commander] RTL switch changed and ON!
0:18:19 WARNING [commander] RTL switch changed and ON!
0:18:36 WARNING [commander] RTL switch changed and ON!
0:20:03 WARNING [commander] RTL switch changed and ON!
0:20:03 WARNING [commander] RTL switch changed and ON!
```

IMHO px4 recognize this RTL command but do nothing for unknown reason

svpcom, 2017-07-30:
@LorenzMeier @dagar 
I've had another flight and RTL failed on high altitude, but succeed when i manually descent and retry it near ground.  PX4 version is v1.6.5. See log:
http://logs.px4.io/plot_app?log=cbcd8c86-2923-4caf-82ab-6d736325b64a

closed 2018-08-01 by santiago3dr