PX4/PX4-Autopilot #7448: Create a new Custom MAVLink Messages , but can not receive it

lhc610github, 2017-06-20:
Hi all,

I try to create a custom mavlink message as the [tutorial](https://dev.px4.io/en/middleware/mavlink.html) told me.  
 1. add the headers and create a new class in [mavlink_receiver.cpp](https://github.com/lhc610github/Firmware/blob/new_test1/src/modules/mavlink/mavlink_receiver.cpp)
 2. make sure it is called in `MavlinkReceiver::handle_message()`

the MAVLink to UART interface is [here](https://github.com/lhc610github/c_uart_interface_example/tree/new_test1)

but still now I can not receive the new message. 
I am not familiar with MAVlink. It seems like the bad crc messages don't pass to handlers. 
how to fix this problem?

LorenzMeier, 2017-08-20:
Sorry, we do not have enough bandwidth to provide user support.

closed 2017-08-20 by LorenzMeier

xdwgood, 2017-10-28:
@LorenzMeier  I think this tutorial is hard to understand, at least I failed to create mavlink messages.
http://qgroundcontrol.org/mavlink/create_new_mavlink_message