PX4/PX4-Autopilot #7449: mc_attitude_control not using integrated gyro data

jgoppert, 2017-06-20:
Why is mc_att_control using the sensor_gyro data topic directly.  I thought I had tracked down the filtering pipeline for the imu data here, https://github.com/PX4/Firmware/issues/7438, only to find that the attitude controller is not using sensor_combined data.

@priseborough  Can we move the sensor calibration and other logic you are doing here into the sensor_combined topic? What was your logic for putting this into mc_att_control?

It also looks like this is not using the integrated gyro data, so it is using:

* current gyro data pipeline to control:
  * 8 kHz raw data on chip
  * low pass filter at 41 Hz on chip
  * sampling at 1 kHz
  * low pass filtered in driver at 30 Hz
  * published to motors at 280 Hz

* recommended gyro data pipeline to control (no FIFO):
  * 8 kHz raw data on chip
  * low pass filter at 250 Hz on chip (highest setting from mpu6000)
  * sampling at 1 kHz
  * integration, output at 280 Hz
  * published to motors at 280 Hz

* recommended gyro data pipeline to control (no FIFO):
  * 8 kHz raw data on chip
  * no on chip low pass filter
  * sampling at 8 kHz
  * integration, output at 280 Hz
  * published to motors at 280 Hz

LorenzMeier, 2017-06-20:
@jgoppert It would be a step backwards to re-integrate into sensor combined. The idea is to have zero latency between gyro data and attitude rate loop. The way the system is currently built allows that.

jgoppert, 2017-06-20:
@LorenzMeier I get the goal, but at minimum shouldn't we be using the sensor_gyro.x_integrated? And why are we filtering it at 41 Hz and 30 Hz currently. That introduces a lot of lag.

LorenzMeier, 2017-06-20:
That is mostly historical based on performance tests we did. I think modern systems are much better matched and it is indeed timely to revisit it.

priseborough, 2017-06-20:
James, the chip samples before applying the internal DLPF, so the DLPF does not help with aliasing

We should not be using integrated quantities for motor feedback becasue that is using a sub-optimal FIR filtering strategy wrt the phase lag vs noise tradeoff and will do a poor job of filtering out noise below 125Hz which is close to the motor fundamental for many systems The most phase efficient way to filter motor vibration noise is a IIR filter, and it should be applied to the raw gyro/accel data before it is downsampled.

jgoppert, 2017-06-21:
My assumption about the sampling was that they were always sampling at 8 kHz internally and then applying a dlpf on those samples before outputting it at 1 kHz, in that case the dlpf would help correct? Is this not what happens? How do we check?

priseborough, 2017-06-21:
See the table on page 13 of this document https://cdn.sparkfun.com/assets/learn_tutorials/5/5/0/MPU-9250-Register-Map.pdf

priseborough, 2017-06-21:
It also gives useful information on group delay for the various filter options.

priseborough, 2017-06-23:
Unless we use the FIFO, we cannot use modes that run the chips internal ADC at faster than 1kHz. Looking at the data it is is likely the chip does not have enough computing resources to run a higher data rate and perform filtering.

nicolaerosia, 2017-06-26:
I will try it on Raspberry Pi 3 with Navio2.
Could you tell me what else I need to change besides the driver providing samples at 8KHz ?
There is a difference between NuttX and DF MPU9250, in NuttX the Integrator is set to reset and in DF is not.
Any idea why is this set this way?
https://github.com/PX4/Firmware/blob/master/src/drivers/mpu9250/mpu9250.cpp#L158-L159
https://github.com/PX4/Firmware/blob/master/src/platforms/posix/drivers/df_mpu9250_wrapper/df_mpu9250_wrapper.cpp#L201-L202

nicolaerosia, 2017-06-26:
LE2: Actually this seems worse than I thought, since the integrator is initialized without a reset period, the integral will only be set on reset ( https://github.com/PX4/Firmware/blob/master/src/drivers/device/integrator.h#L53-L63 ).
The DF MPU9250 uses this value to report the x/y/z_integral, which logically should be zero ( https://github.com/PX4/Firmware/blob/master/src/platforms/posix/drivers/df_mpu9250_wrapper/df_mpu9250_wrapper.cpp#L623-L629 ).

nicolaerosia, 2017-06-26:
Related issue: https://github.com/PX4/DriverFramework/issues/209

priseborough, 2017-06-26:
The requirement to publish the raw accel or gyro data as received from the FIFO would be driven by a specialised data logging mode for vibration analysis and should have it's own topic that is only published when the logging mode is active. Otherwise there is no point publishing faster than the rate loops require to be updated.

nicolaerosia, 2017-06-26:
@priseborough wouldn't EKF benefit from publishing faster? Is there a table with the rate loops frequency?

priseborough, 2017-06-26:
@nicolaerosia No, because it only runs the output predictor at the sensor_combined rate. Delta angles and velocities received in the sensor_combined topic are integrated across 12msec time steps using a full inertial nav calculation and stored in a FIFO buffer for use by the EKF prediction and correction cycle which runs ever 12msec on a delayed time horizon.

The main requirement is that the EKF receives the delta angle and velocity data no slower than 12msec the update rate of the attitude, velocity and position controllers or 1/ 12 msec whichever is the faster.

In theory this will introduce some additional coning and skulling error if high frequency vibration is present, but currently the driver performs  compensation for coning errors on the IMU data before delta angles are downsampled and published on the sensor-gyro message, and skulling errors are swamped by other errors in our low grade sensors.

nicolaerosia, 2017-06-26:
@priseborough I see, thank you. So the right way to approach this would be to do filtering for a BW ~100 Hz (1000/12 ~ 83.33 Hz), use Integrator to integrate data before it's time to publish data that's going to end up in sensor_combined ?
If the EKF runs at 84 Hz, then why the current driver publishes at a higher rate?

priseborough, 2017-06-26:
No digital noise filtering is required for high rate data before integration in the driver. That integrated delta angle and velocity data is for inertial nav and should only be used by the EKF. We should have separate IIR filtered rate and accel data for the control loops.

The EKF uses an output observer that provides kinematic state outputs at the same rate as the sensor combined topic. The output observer  tracks the EKF states at the delayed time horizon and predicts forward to the current time horizon. This avoids the unnecessary computational and storage cost of running the EKF  at the same rate as the control loops (would require large data buffers).

closed 2017-06-29 by jgoppert