PX4/PX4-Autopilot #7450: Multicopter: Anyway to directly control MAIN OUT from modified attitude_control code?

NickSadjoli, 2017-06-20:
Hello! I am relatively new to the PX4 scene, and thus I am still quite confused on what can and cannot work, so any help would be appreciated.

So I am currently trying to develop a '_flying car_' using the PX4 flight controller + firmware. Initially I thought that creating a new airframe and making a custom mixer should do the trick. However, it seems that doing so is not achieving what I want.

Specifically, I am trying to modify the control scheme that does the following when I choose the **ACRO** mode:

1. When I put an RC throttle stick **under** a certain value (let's say 1/4 of the total stick), the controller would activate the 5th and 6th MAIN OUT pins (car motors would be on these pins), and output **pwm values** based on the **Roll and Pitch RC inputs** (basically **mapping** roll and pitch inputs to the MAIN 5 and 6 OUT pins)

2. However, when the throttle stick is **above** the value, the controls would return to a **normal Multicopter control**, and the 5th and 6th MAIN Pins would be disabled, or having PWM_Min output.

I have tried to achieve this by creating a **custom mixer** configuration and also adding an attitude/position control scheme that is basically a **modified mc_att_controller** code that is available in one of the src/modules/ directory. In particular, the following is my custom mixer:

(I am trying to map pins 5 and 6 from MAIN to actuators 4 and 7 from control group 0)
`R: 4x 10000 10000 10000 0

M: 1
O:      10000  10000      0 -10000  10000
S: 0 4  10000  10000      0 -10000  10000

M: 1
O:      10000  10000      0 -10000  10000
S: 0 7  10000  10000      0 -10000  10000
`

And here is the acro/rate_control portion of the code that I am currently modifying from the mc_att_control code:
`
if (_v_control_mode.flag_control_rates_enabled) { //entering ACRO mode

				/* publish actuator controls */
				_actuators.control[0] = (PX4_ISFINITE(_att_control(0))) ? _att_control(0) : 0.0f;
				_actuators.control[1] = (PX4_ISFINITE(_att_control(1))) ? _att_control(1) : 0.0f;
				_actuators.control[2] = (PX4_ISFINITE(_att_control(2))) ? _att_control(2) : 0.0f;
				_actuators.control[3] = (PX4_ISFINITE(_thrust_sp)) ? _thrust_sp : 0.0f;

				if (_thrust_sp < _params.throttle_thres){
                                       //if throttle stick or vehicle throttle is below certain value, update actuators 4 and 7 of group 0 (attitude control group)
					_actuators.control[4] = _actuators.control[0];
					_actuators.control[7] = _actuators.control[1];
				}
				else{
                                      //if not, actuators 4 and 7 are 0f
					_actuators.control[4] = 0.0f;
					_actuators.control[7] = 0.0f;
				}

				_actuators.timestamp = hrt_absolute_time();
				_actuators.timestamp_sample = _ctrl_state.timestamp;

				_controller_status.roll_rate_integ = _rates_int(0);
				_controller_status.pitch_rate_integ = _rates_int(1);
				_controller_status.yaw_rate_integ = _rates_int(2);
				_controller_status.timestamp = hrt_absolute_time();

				if (!_actuators_0_circuit_breaker_enabled) {
					if (_actuators_0_pub != nullptr) {

						orb_publish(_actuators_id, _actuators_0_pub, &_actuators);
						perf_end(_controller_latency_perf);

					} else if (_actuators_id) {
						_actuators_0_pub = orb_advertise(_actuators_id, &_actuators);
					}

				}

				/* publish controller status */
				if (_controller_status_pub != nullptr) {
					orb_publish(ORB_ID(mc_att_ctrl_status), _controller_status_pub, &_controller_status);

				} else {
					_controller_status_pub = orb_advertise(ORB_ID(mc_att_ctrl_status), &_controller_status);
				}
			}
`

So far I have tested that the mixer files have a good effect on the controller (when I map the pins to actuators 0, 1, 2, or 3 [roll,pitch,yaw,thrust] from control group 0, it was behaving according to the inputs from RC). However, I cannot seem to make flight code to actually change the actuator_control value like I want.

Or rather, I cannot seem to make the code to do **directly control** the MAIN OUT pins outputs. I have tried to use different **uORB messages** to try to invoke this behaviour, but I am still not getting the result.

Is this because of the separation between the mixer and actuator logic that is part of the PX4 architecture? Are there any suggestions for me on trying to achieve the control scheme that I described above?

In summary: I am trying to make it such that when RC Throt < certain value, MAIN OUT 5 and 6 would be mapped to RC Roll/Pitch, and when RC Throt >= certain value, it will act like normal Multicopter in ACRO mode. If possible all MC attitude characteristics would still be preserved during this process.

Looking forward to suggestions and help, thank you!

dagar, 2017-06-20:
Perhaps I'm not following what you're looking for, but do you want manual input (RC) mapped to these outputs? If so you should be copying from _manual_control_sp.

NickSadjoli, 2017-06-20:
Yes, what I want is to have the input from RC Roll and Pitch to be **programmatically** mapped to the MAIN output pins, when RC Throttle stick value is below a certain point. Is _manual_control_sp still the way to go then?

Thanks for the reply!

NickSadjoli, 2017-06-20:
Also, would my method of mapping the MAIN OUTPUT 5 and 6 to actuators 4 and 7 enable them to change pwm outputs when actuators 4 and 7's valuesare changed?

If so, is using _actuator_controls the correct way to set the flaps/gears actuator (actuator group 0, number 4 and 7) to a certain value? 

I hope to hear from you again soon, and so so sorry for so many questions like this, since I am very new to the PX4 project.

dagar, 2017-06-28:
Again still not sure I completely follow you, but it sounds like _manual_control_sp is the way to go. Yes you can use flaps/gear on actuator_controls, but they need to be in your mixer.

Let's move this type of discussion to http://discuss.px4.io/ or gitter and save github for bug reports and feature requests.

closed 2017-06-28 by dagar