PX4/PX4-Autopilot #7455: Pixhawk with PX4 firmware - No way to configure retractible landing gear?

hawkeyeuas, 2017-06-21:
I am running a standard X copter configuration, formerly running Ardupilot on the Pixhawk.

I was advised to change to PX4 so that offboard mode could easily be configured.  When I did, I discovered that there is no way, at least that I could find, to reconfigure my retractable landing gear.

The landing gear not working with PX4 is a huge issue.

I don't have any problems configuring the offboard mode on the PX4 firmware, but don't get any landing gear parameters described in any online narratives, except for Ardupilot.

When I switch to Ardupilot, I get the landing gear parameters, but can't find the parms to set up the offboard mode.

A catch 22.   Can anyone point me to a narrative to deal with offboard setup on Ardupilot?

Thanks

LorenzMeier, 2017-06-22:
Of course there is an option. You need to configure a few things:

  * Gear switch: `RC_MAP_GEAR_SW` - set it to the RC channel you want it to be on
  * You need a gear mixer to pass the gear channel to the AUX outputs:
```
#
# This maps actuator_controls_0[7] (gear) to the first PWM output
#

M: 1
O:      10000  10000      0 -10000  10000
S: 0 7  10000  10000      0 -10000  10000
```
  * The gear mixer needs to be loaded:
```
#
# Load the gear mixer onto fmu
#
mixer load /dev/px4fmu /etc/mixers/gear.mix
```

All this can of course be automated - you would need to extend your startup script to include the right config.
https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/init.d/4011_dji_f450

I can do a PR later today with that all enabled for you by default. The gear is also operated in auto modes accordingly already.

hawkeyeuas, 2017-06-22:
Lorenz,

Thank you so much for your prompt reply and assistance.  I decided to re-install the Ardupilot firmware and reconfigure the landing gear functionality.

I think I mentioned that the reason for the change to PX4 was to set up offboard mode for a companion Odroid computer.  I was told that the terminology is different on Ardupilot and is called guided mode.  I have it setup, but am unsure if it was done correctly since QGroundControl seems to be a little confused at this point.  I arms and disarms on its own when booted up saying it has an "unknown" flight mode.

Could you shed some light on this subject?

I will need to reconfigure the gimbal and camera functionality next.  Since I normally purchase the pre-configured systems, this is a learning experience for me.

Could you give me a brief tutorial on how to best set up the gimbal and camera on Ardupilot, please ?

Thanks again,

Abe Piemme
Hawkeye UAS

www.hawkeyeuas.com<http://www.hawkeyeuas.com/>

706.847.6001 voice
706.677.1565 fax

<https://www.facebook.com/hawkeye.uas.9>LIKE us on facebook<https://www.facebook.com/hawkeye.uas.9>

<https://www.youtube.com/channel/UC4xHck0OiXETWkqj8xJo6LA>

Privileged or confidential information may be contained in this message.  If you are not the addressee indicated in this message, you may not read it, copy it or forward it to anyone.

If this message has been received in error, you should destroy this message and notify us immediately. Thank you

https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/init.d/4011_dji_f450

LorenzMeier, 2017-06-22:
This is for PX4 firmware issues - not for general user support.

closed 2017-06-22 by LorenzMeier

dharma-kc, 2017-11-07:
Dear Lorenz,

Could you suggest me where do I put this command:
mixer load /dev/px4fmu /etc/mixers/gear.mix

If I put the gear.mix file on the microsd card in above format, it will automatically be loaded before corresponding default mix file, right? Do I need to load it again ? If we need to load, where do I place this load command?

Thank you, 
Regards.

frontw, 2019-10-03:
How to test, that it works. Is it possible to do at table, not in flight?
I listen to actuator_control uorb, and landing gear is always -1, not depending corresponding radio channel.