PX4/PX4-Autopilot #7457: Motor spinning on USB power reset

lericson, 2017-06-22:
Despite the fixes in #7123, we still get motors spinning on unplugging the USB cable.

**Steps to reproduce**:
1. Plug in battery, USB cable
2. Start QGroundControl
3. Unplug USB cable

It only happens if QGroundControl has been talking to the PX4, as this triggers USB telemetry boolean that actually makes the PX4 reset after USB has been unplugged.

I have looked at the PWM signal with an oscilloscope and have pictures of it, but they aren't very telling.

**Workaround:**
In `src/modules/commander/commander.cpp`, find the `usleep()` just before the reboot and put `board_on_reset(0);` *before* the sleep. This sets all PWMs to zero.

A better long term fix would be to add a sleep to the `board_on_reset`.

lericson, 2017-06-22:
We have the PixRacer hardware version 4, by the way.

davids5, 2017-06-22:
@lericson - Thank you for reporting this. Please test https://github.com/PX4/Firmware/pull/7459

For a faster test cycle time: A `reboot` command on the debug port will also trigger the spin. 
It is tricky to catch. The asynchronous reset has to occur when the PWM pins are high. So more testing is better, testing with a logic analyzer with analog, is the best way to see the fix is solid. 

On the HW I tested the discharge rate was fast enough. I had assumed that is not the same on all HW and had the hook in place just incase. The PR use it.

lericson, 2017-06-22:
It happens very consistently for us. Will try the PR on Monday, thanks! I had tried usleep at the same location but it didn't work -- might've been to short an interval too.

davids5, 2017-06-22:
@lericson  
Thank you! Please let me know how the testing goes. I may also need more details the proposed fix you have and the HW attached.

ndepal, 2017-06-27:
Just happened to me as well with a PixRacer on master ca480ff.

davids5, 2017-06-27:
@ndepal Would you please test https://github.com/PX4/Firmware/pull/7459 and see if it stops the issue.

ndepal, 2017-06-27:
Just tested #7459, it does not fix the issue.

davids5, 2017-06-27:
@ndepal would you be able to do a few more tests?

1) Comment out the usleep found [here](https://github.com/PX4/Firmware/blob/9c25f93d09fd57387c35ae02688d20575a8ef027/src/modules/commander/commander.cpp#L2052) 

Does the motors still spin?

2) Add the sleep back in as `usleep(10000)`

Does the motors still spin?

These are just tests to try to understand where the PWM is getting stretched.

ndepal, 2017-06-28:
Tested as requested.

  1. motors still spin
  2. motors still spin

davids5, 2017-06-28:
@ndepal Thank you for testing.  

Would you mind one more test?

Do as @lericson suggested place the  `board_on_reset(0);` call prior to [this line](https://github.com/PX4/Firmware/blob/9c25f93d09fd57387c35ae02688d20575a8ef027/src/modules/commander/commander.cpp#L2052)

I will need to duplicate your setup. Please list what you have connected to the PixRacer and on what connector. Part number or links would be appreciated.

ndepal, 2017-06-29:
`board_on_reset(0);` fixes the problem for me, but only with a long enough sleep afterwards. I did not get spinning motors with `usleep(300000);` or above, but I did with `usleep(200000);` and below.

My setup:
Pixracer from auav.co
ESCs: [Favourite Little Bee 30A](https://hobbyking.com/en_us/favourite-little-bee-30a-2-6s-esc-no-bec.html?___store=en_us)
Motors: [Emax cooling MT2206 II 1900KV](http://www.helipal.com/emax-mt2206-v2-mini-cooling-motor-1900kv-ccw.html)
nothing else attached for these tests.

davids5, 2017-06-29:
@ndepal Thank you that is super helpful. I think I understand what is happening. See https://github.com/PX4/Firmware/pull/7459/commits/0484ee32239a9534db455b0cf3788440dfa3b92a for the details. I have updated https://github.com/PX4/Firmware/pull/7459. would you please fetch it again and test it with no other modification.

ndepal, 2017-06-29:
Pulled and re-tested #7459, the issue is fixed for me.

davids5, 2017-06-29:
@ndepal - thank you!

@lericson  I have updated #7459. would you please fetch it again and test it with no other modification.

LorenzMeier, 2017-07-09:
Should be fixed.

closed 2017-07-09 by LorenzMeier

lericson, 2017-07-17:
Can confirm it no longer happens. On a related note, we got the living crap scared out of us when our hexacopter started running its rotors full speed due to this. It even took off from the desk where it was seated, and not on USB reset either -- a simple reboot.