PX4/PX4-Autopilot #7458: Add landing gear tutorial and default instructions

LorenzMeier, 2017-06-22:
* Gear switch: `RC_MAP_GEAR_SW` - set it to the RC channel you want it to be on
  * You need a gear mixer to pass the gear channel to the AUX outputs:
```
#
# This maps actuator_controls_0[7] (gear) to the first PWM output
#

M: 1
O:      10000  10000      0 -10000  10000
S: 0 7  10000  10000      0 -10000  10000
```
  * The gear mixer needs to be loaded:
```
#
# Load the gear mixer onto fmu
#
mixer load /dev/px4fmu /etc/mixers/gear.mix
```

dagar, 2017-06-22:
Longer term could we find a way to do this without having to deal with a mixer?

I'm also wondering if it would make sense to move things like landing gear to a separate control group and handle it outside of the attitude controllers. That module could have landing gear config via param and would be accessible for both MC and FW.

matanhavi, 2017-06-24:
Is there way to control the landing gear by mavlink?
Does they controlled by logic? for example, during RTL, does it goes down automatically?

dagar, 2017-06-26:
@matanhavi for a multicopter yes it will go down automatically.

matanhavi, 2017-06-26:
@dagar Thanks, I will try it.
Does it have option to be controlled by mavlink?
and/or how they goes up?
I tried to find the related code, to follow the logic, without success.

dagar, 2017-06-26:
Here for AUTO - https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/mc_pos_control_main.cpp#L1560

mattgorski, 2018-08-20:
I would like to see more documentation on how to automate this when using gamepad/joystick inplace of RC.  Would I just load the mixer and also what is the proper way to load this mixer put on microsd in /etc/mixer does this need a config.txt?  Thanks in advance.

hamishwillee, 2018-08-21:
Reopening - the tutorial does not exist, so this work has not been done. Probably this should be in the PX4 user guide or devguide issue tracker. 

@dagar 
1. Above you suggested that we should improve the system so that mixer not required (YES). Did this happen? We have `RC_MAP_GEAR_SW` to map channel to gear switch, but I envisage we're talking a second parameter to map RC_MAP_GEAR_SW to a specific MAIN or AUX (or could this be on some other bus - ie UAVCAN?)
1. You also suggested factoring this so that it could be used on all vehicle types - did that happen? So auto landing not supported on VTOL or FW?
1. Generally you'd add the mixer etc as per http://dev.px4.io/en/advanced/system_startup.html#customizing-the-system-startup ?
1. #10279 asks about Joystick button mapping. Do you happen to know if there is any equivalent to RC_MAP_GEAR_SW for mapping joystick buttons. Certainly this is not supported via the UI: https://docs.qgroundcontrol.com/en/SetupView/Joystick.html  (or is this a "Don/Gus" question?)

mattgorski, 2018-08-30:
I am going to assume that we need to add a config.txt with the following correct (along with gear.mix above)?

`set MIXER_AUX gear
set PWM_AUX_OUT 1234
set PWM_AUX_DISARMED 1500
set PWM_AUX_MIN 1000
set PWM_AUX_MAX 2000
set PWM_AUX_RATE 50`

dagar, 2018-08-31:
The portion that applies to multicopter and RC_MAP_GEAR_SW is in reasonably good shape, but FW and joystick button mapping is still missing.

hamishwillee, 2019-01-31:
@dagar I assume you still want to improve the architecture around this as per https://github.com/PX4/Firmware/issues/7458#issuecomment-310419573

Otherwise, if we want to document this as it is now there is not enough info - see my post: https://github.com/PX4/Firmware/issues/7458#issuecomment-414621313

closed 2019-02-14 by stale[bot]