PX4/PX4-Autopilot #7464: PWM Camera trigger does not work X2.1

tubeme, 2017-06-23:
Hello guys,

We've got a pure PWM controlled camera and we tried to connect it with the 1.6.3. We set up like this:

TRIG_MODE 4
TRIG_INTERFACE 4
TRIG_PINS: 56

We connected the camera to pin AUX 5 and then tried to manually trigger the camera from the info panel and from Camera Settings panel as well. Neither one worked. Tried the camera with a regular PWM RX. With a 3 position switch it works perfect. We tried AUX 6 as well. And AUX1. We used the default profile of Delta Quad VTOL. Then we connected a servo to the AUX5 and 6 and again manually tried to triger the servo this time. Nothing happened. Is there something in the mixer we are missing?? Could not find the reason for not sending US to the servo.

We could not find a way to setup our camera. Here how it works.

Pulse Signal:

[image]

Level 0: For video recording

Leve 1: Do nothing

Level 2: Taking Photo

How does the camera to recognize the signal:

Level 0 > 1 > 2, stop recording, take a photo
Level 2 > 1 >2, take a photo
Level 2 > 1 > 0, start recording

We saw that the Seagull #MAP2 trigger also had a settings that we could not find a way to setup?

[image]

tubeme, 2017-06-23:
Another related issue was that in QGC in the Camera Trigger control panel the default for TRIG_INTERFACE was shown as "Unassigned: 4". Then if we switch to let say "Seagull MAP2" and restart, we could not see the Unassigned:4 option anymore. We have to forcefully change trig_interface = 4.

Also there is something with the color scheme recently. Everything in the settings drop down boxes is white letters over 10-20% grey background. I have to stare at the screen to see the options.

mhkabir, 2017-06-24:
[image]

Works for me on v1.6.3 and a Pixhawk. This might be something specific to the X2.1, but I don't have that hardware to debug.

tubeme, 2017-06-24:
@mhkabir Did you connect something to the AUX 5?

tubeme, 2017-06-25:
@mhkabir It could not be related to X2.1. This is tested on Pixhawk FMUv2. 1.6.3

tubeme, 2017-06-26:
I dismissed the X2.1 from the name of the issue because all the tests we do are with plain Pixhawk. We still cannot make it work. The servo Aux5,6 are dead in our tests, nothing happens on the pins. We tried connecting plain servo, powering the servo bus with bec, we tried the PWM camera as well. We tried moving to AUX1,2 with the same results. Nothing happens. I must say we try triggering the camera only with the manual buttons in the instrument panel and in the Camera Triger settings panel.

[image]

LorenzMeier, 2017-06-26:
I know its working as I just got a long survey log from 1.6.3. I'm currently trying to identify what your setup issue is. Can you please try TRIG_INTERFACE = 2 instead?

tubeme, 2017-07-07:
With HKPilot32 FMUv2 it is not working on the ground with manual trigger buttons.

We tried another HKPilot32 and Pixhawk Mini as well with the same results.

We tried all combinations:

Generic PWM. SeagulMap2 as well. We tried all the options for trigger mode as well.

Tried ch5 ch6 and ch1 and ch2 as well.

[image]

[image]

With servo connected to all the channels nothings come out from the servo bus.

We are suppose to capture manually first then fly test it correct?

@LorenzMeier Could it be that the manual trigger only does not work and in the mission it will be perfectly fine?

tubeme, 2017-07-11:
OK we have development. It was a cascading failure of servos and power that led us to believe we have no PWM output. In the configuration PWM and SeagulMAP2 it outputs PWM signal to the AUX 5 ch. 

But we still cannot trigger our camera. Here is the problem.

By default the Aux bus is in 1500us when armed so the servo stays in the middle. When we trigger we see very little movement for the servo. Like 200-250us, which is not enough to trigger our camera and is not with the same logic. To take a picture we need to stay in 2000us then make a pulse to 1500us and then back to 2000us. 

To start video recording we need to stay in 2000us then go to 1000us the camera starts recording.

Then if we move from 1000us to 2000us the camera stops recording.

We did not find a way to control this behavior except to make the ch5 armed state to be 2000us with the script. 

Also it is not a bad idea to have another video trigger in the test panel as well as the picture trigger.

Probably we could use the Camera MAV commands from the Mission items...

mhkabir, 2017-07-11:
The default "Trigger On" pulse is hard-coded to 1700us. This can be made a parameter for flexibility and exposed via QGC's Camera Trigger UI. 

> To take a picture we need to stay in 2000us then make a pulse to 1500us and then back to 2000us. To start video recording we need to stay in 2000us then go to 1000us the camera starts recording.

This will require an interface specific to your camera, and not something generic. Have a look at the MAP2 interface to see how to implement the low-level camera-specific pulses, etc.

closed 2018-02-21 by PX4BuildBot