PX4/PX4-Autopilot #7465: Using sonar MB1242 in LPE

lyhsteven, 2017-06-24:
Hello guys,
We want to use sonar MB1242 to measure the altitude in LPE. To make it happened, I've already done these:
1. I uncomment the drivers/mb12xx in cmake/configs/nuttx_px4fmu-v2_default.cmake
2. I turn the param SENS_EN_MB12XX from 0 to 1 in src/modules/sensors/sensor_params.c and also in QGC
3. I don't know how much to set LPE_FUSION for I set it 174
But it still doesn't work.
Could anyone  give me some advice, I'll be appreciate to that.

deksprime, 2017-06-26:
Which interface are you using for  the sonar to the FCU? Maybe you are trying to use an interface which is not supported in the driver.

lyhsteven, 2017-06-27:
I got it. It also needed to edit in rcs that turn **if param compare SENS_EN_MB12XX 1**, then it will work.

closed 2017-06-27 by lyhsteven