PX4/PX4-Autopilot #7467: Setting mode to PosCtl after RTL causes large velocity setpoint

nkhoit, 2017-06-24:
If you command an RTL then set mode to Manual PosCtl, the vehicle will 

SITL Log: http://logs.px4.io/plot_app?log=dd5b9854-3c7a-46ce-b97d-457dd3185a88
Replication steps: 
- Take off in PosCtl
- Command RTL
- Set mode to PosCtl
The vehicle will quickly ramp up to 9 m/s with no stick input then slow down after a few seconds
It's a lot clearer to see if you set all the VEL_MAX parameters to 3 m/s
So I set 
MPC_VEL_MAX_MAX = 3 m/s
MPC_XY_CRUISE = 3 m/s
MPC_XY_VEL_MAX = 3 m/s
You'll see the vehicle go very slowly at 3 m/s then when you set mode to POSCTL, it'll quickly shoot up to 8-9 m/s ground speed.

A plot of the local_position vel_y and setpoint vel_y:
[image]

MaEtUgR, 2017-06-26:
Interesting finding! This looks to me a like a slewrate being applied without having reset the previous setpoint on which the calculation is based.

But if I see it correctly you do not fly fast before switching to RTL which leaves the open question where this very high previous velocity setpoint comes from.

@Stifael any recent changes that might explain?

Stifael, 2017-06-26:
@MaEtUgR I commented here:
mc_pos_control: Properly constrain xy velocity setpoints #7468

nkhoit, 2017-07-09:
Fixed in #7538

closed 2017-07-09 by nkhoit