PX4/PX4-Autopilot #7474: Make airframe outputs consistent

hamishwillee, 2017-06-26:
It would be good if the airframe outputs could use consistent outputs for the same function.

So for example, the [standard plane](https://docs.px4.io/en/airframe_reference.html#standard-plane) frames use MAIN3, 4, or 5 for throttle.

(FYI I have updated the user guide so that the airframe reference is now the canonical place to get the output information, but I still think consistency in setup would make things easier for users).

dagar, 2017-06-26:
There will be exceptions of course, but we can definitely do a better job being consistent.

LorenzMeier, 2017-07-09:
Done.

hamishwillee, 2017-07-09:
@LorenzMeier Please reopen this. The change I made better organises the content, but (despite my comment in #7482 does not "fix" the problem).

To fix this requires an update to the configuration files for the specific vehicles (mixers?) to make sure that they map the same output for throttle (and where possible other outputs). As I understand it, the change is technically easy to make, but would be a considerable "compatibility break."

closed 2018-02-21 by PX4BuildBot