PX4/PX4-Autopilot #7476: Pixhawk compass won't calibrate

Kaiserollz89, 2017-06-26:
I am having a problem getting my compass to calibrate in qgroundcontrol. I am using a Pixhawk Mini flashed with the px4 firmware. I didnt have any issues calibrating the gyro, accelerometer, or level horizon. But when I attempt to calibrate the compass I get an error message " Mag 0(ID 26178) no matching uORB devid... calibration failed: mag".

I have tried reflashing the firmware, calibrating with and without peripherals plugged in, restarting QGroundControl, and even starting my computer, and just starting the calibration process from the beginning.Ive looked through other discussion boards but haven't found anyone having this problem. Any suggestions???

ndepal, 2017-06-30:
I am seeing the same with a PixRacer flashed with the current release. It happens with and without an external GPS/mag attached.

@mhkabir Your comment on the [discuss thread](http://discuss.px4.io/t/pixhawk-compass-wont-calibrate/3702/2) sounds like you have seen this before/know what the problem is?

mhkabir, 2017-06-30:
@ndepal can you try this branch? https://github.com/PX4/Firmware/pull/7434

ndepal, 2017-07-03:
@mhkabir tried that branch, but it doesn't fix the problem for me, at least not reliably. After first flashing I was able to start calibrating (through wifi telemetry) the compass. I canceled the calibration, power cycled, and tried again. Then I got the same error message again.

ndepal, 2017-07-06:
I have been trying to reproduce this and I believe I have found the cause for the issues I am having. The PixRacer in question is in a prototype frame we have 3d printed for a proof of concept test. This frame has a cover element that we are holding in place with magnets to quickly and easily attach and detach it.

Of course, magnets near a magnetometer are not the best idea. I only wanted to fly manually and therefore figured I wouldn't have to worry about accurate yaw estimation.

It appears now that the magnetic field from the magnets not only distort the magnetic field causing false yaw estimates, but they actually cause the mag sensor not to produce any measurements, giving `Mag #0 fail:  TOUT!` (visible only in the `msgs.txt` on the SD card, not in QGC) messages and visible gaps in the logs. 

This rather confusingly manifests itself during the calibration with the `Mag 0(ID 26178) no matching uORB devid` message, rather than, say, a message alerting me that the mag is timing out or saturating (if that's the cause for the timeout).

Here is a log where I place the Pixracer into a rectangle of magnets and remove it again several times. You can see the drop outs in the mag measurements in `sensor_combined_0.magnetometer_timestamp_relative`.
http://logs.px4.io/plot_app?log=efe58a2d-d088-4e08-9e52-39f31857e8ce

silverburn, 2018-02-11:
I have just experienced this problem. Was able to calibrate in Arduplane, but cannot calibrate in Qcontrol. Bring up the Accelerometer screen (the plane in red boxes), then fails a few seconds later with:

Mag 0 (ID 73225) no matching uORB devid.

Multiple attempts at firmware and parameter reset. No change.

Plane is a TBS Cappy 2.

dagar, 2018-02-11:
@silverburn Do you have an external compass? How are you powering the autopilot? How is it connected to QGC?

silverburn, 2018-02-11:
Yes, External and internal.

Update: May have solved it. Moved external away from a servo, and now it calibrates. Large offsets on internal compass (I cant move that away from the servo), so I have disabled it.

closed 2018-08-01 by santiago3dr