PX4/PX4-Autopilot #7479: Px4 <==> Piksi Multi RTK communication

hitimo, 2017-06-26:
Has somebody worked / is working on the SBP2 driver that 
- reads navigation solution from and 
- sends correction data to Piksi Multi?

This is implemented for ArduPilot:
https://github.com/ArduPilot/ardupilot/blob/3018728f50e5a8f5f73539da8c74f26393fbf286/libraries/AP_GPS/AP_GPS_SBP2.cpp
https://support.swiftnav.com/customer/en/portal/articles/2803651-piksi-multi-ardupilot-integration-guide

[image]

bkueng, 2017-06-27:
Not that I know of... but we happily accept PR's :)
We currently support the u-blox RTK:
https://docs.px4.io/en/advanced_features/rtk-gps.html
https://dev.px4.io/en/tutorials/rtk-gps.html

LorenzMeier, 2017-06-27:
Let me chime in here:

  * We support the Ashtech RTK GPS which uses NMEA streams
  * We support forwarding RTCM correction streams from QGroundControl to PX4

So 90% of what you need is likely already there.

hitimo, 2017-06-27:
Thanks for the pointers. My corresponding PR will (hopefully) follow soon then.

TarekTaha, 2017-07-02:
@hitimo we tried to use Swift RTK re-route via mavros in the past (you will have to enable hil_gps to do so), and it was not as reliable as we hoped.

https://github.com/kuri-kustar/mavros/blob/master/mavros_extras/scripts/swift_rtk_reroute.py
https://github.com/kuri-kustar/mavros/blob/master/mavros_extras/src/plugins/gps_reroute.cpp

This is not the best approach, as @LorenzMeier mentioned, the NMEA support should be enough, but swift have limited support for NMEA.

If you manage to write a driver for it, then I would be happy to test it.

devbharat, 2017-11-01:
@TarekTaha can you add a little detail in what way you found Piksi to be unreliable? Was it just unreliable when used for RTK based position control (I could imagine rerouting causing delay, maybe robustness issues), or something more low level that you also effect it's PPK accuracy?

TarekTaha, 2017-11-06:
@devbharat it wasn't only the re-routing. 
Piksi is very sensitive to electronic noise, and shielding it was quite a challenge, even a camera cable 20 cm close to the unit was causing the signal to drop significantly. 
We insisted on using PX4 flight stack, and not the APM, so we had to create this re-routing hack which did work, but it wasn't ideal.

dagar, 2018-01-23:
@hitimo did you make any progress here?

hitimo, 2018-01-31:
@dagar Unfortunately only little as this had a low prio for us in the last months.

closed 2019-02-13 by stale[bot]