PX4/PX4-Autopilot #7483: Airframes reference - what do the MAIN mean

hamishwillee, 2017-06-27:
https://docs.px4.io/en/airframe_reference.html#simulation

@bkueng Regarding:
[image]

1. Image is plane, but one of the simulations is quad
2. For quad the MAIN values above will be wrong.
3. What do the values for MAIN mean in a simulation?

bkueng, 2017-06-27:
>  Image is plane, but one of the simulations is quad

I suggest we remove the image.

>    For quad the MAIN values above will be wrong.

The outputs should be added here: https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/init.d/1001_rc_quad_x.hil

>    What do the values for MAIN mean in a simulation?

It's the channel for the mavlink message: the index of `controls` in [HIL_ACTUATOR_CONTROLS](http://mavlink.org/messages/common#HIL_ACTUATOR_CONTROLS)

dagar, 2017-06-27:
These might actually be better as a Simulation entry under the regular categories. The number of categories is kind of overwhelming.

[image]

Would we be better off with a much simpler set of groups?
Something like initially selecting the vehicle from maybe 8 generic groups, then on the side you see the specific airframe with metadata. I'm getting pretty off topic for a firmware issue.

hamishwillee, 2017-06-28:
@bkueng We've progressed a little in https://github.com/PX4/Firmware/pull/7482 by splitting the simulation into different "classes" . BUT still need to know how to get rid of the plane image for the quad section - your help appreciated. This item can now be closed.

@dagar In #7482 you will see that for the docs we now have a "class" (e.g. vehicle type like VTOL). This means that as you have suggested the simulation gets split by class to the right place. 

How QGC does this should be raised in QGC right?

> These might actually be better as a Simulation entry under the regular categories.
 
As a user in QGC I think I prefer to have them all grouped together under "Simulation". The problem is that then the icon will be wrong for at least some vehicles.

My reasoning is that I probably want to see what is available when I want to simulate, rather than hunt in vehicle types. I'd feel differently if we had simulation options for every frame.

> Would we be better off with a much simpler set of groups?

Of course, if it can be done reasonably :-) ONe option might be to do as we have in the docs and split to vehicle type first - copter, vtol, plane, boat, other ...

closed 2017-06-28 by hamishwillee

bkueng, 2017-06-28:
> BUT still need to know how to get rid of the plane image for the quad section

It comes from here: https://github.com/PX4/Firmware/blob/master/Tools/px4airframes/srcparser.py#L54 and uses this image: https://github.com/PX4/Devguide/blob/master/assets/airframes/types/AirframeSimulation.svg

> Of course, if it can be done reasonably :-) ONe option might be to do as we have in the docs and split to vehicle type first - copter, vtol, plane, boat, other ...

+1