PX4/PX4-Autopilot #7486: "make px4cannode-v1_bootloader" can't generate .bin file

leomatrix, 2017-06-27:
"make px4cannode-v1_bootloader" can't generate .bin file, so what should we flash the cannode bootloader to stm32f103 ???

davids5, 2017-06-27:
I just built it from master. 

If you can not build - Check your version of arm gcc with `arm-none-eabi-gcc --version` 

It needs to be: 5.4.1 20160609 

It generates an elf file:

build_px4cannode-v1_bootloader/src/firmware/nuttx/firmware_nuttx

That file can be converted to a bin file with this command.

`arm-none-eabi-objcopy -O binary firmware_nuttx firmware_nuttx.bin`

My build details
```
make px4cannode-v1_bootloader
GIT_SUBMODULES_ARE_EVIL is defined - Skipping All submodule checking!
-- CMAKE_MODULE_PATH: mainline/Firmware/cmake
-- Nuttx build for px4cannode-v1 on m3 hardware, using bootloader 
-- Build Type: MinSizeRel
-- PX4 VERSION: v1.6.2-125-g878817c
-- CONFIG: nuttx-px4cannode-v1-bootloader
-- The ASM compiler identification is GNU
-- Found assembler: /usr/lib64/ccache/arm-none-eabi-gcc
-- Found PythonInterp: /usr/bin/python (found version "2.7.5") 
-- Found PY_jinja2: /usr/lib/python2.7/site-packages/jinja2  
-- C compiler: arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 5.4.1 20160609 (release) [ARM/embedded-5-branch revision 237715]
-- C++ compiler: arm-none-eabi-g++ (GNU Tools for ARM Embedded Processors) 5.4.1 20160609 (release) [ARM/embedded-5-branch revision 237715]
-- Configuring done
-- Generating done
-- Build files have been written to: mainline/Firmware/build_px4cannode-v1_bootloader
[49/49] Linking C executable src/firmware/nuttx/firmware_nuttx
```

I am closing this, Please feel free to reopen if this does not work for you.

closed 2017-06-27 by davids5

leomatrix, 2017-06-28:
@davids5  another problem is "make px4cannode-v1_default" was lead to a error:

yang@yang-B85M-D2V:~/Firmware$ make px4cannode-v1_default 
-- CMAKE_MODULE_PATH: /home/yang/Firmware/cmake
-- Nuttx build for px4cannode-v1 on m3 hardware, using nsh 
-- Generating UAVCAN Bootable as org.pixhawk.px4cannode-v1-1.0-0.1.d5eab23f.uavcan.bin
-- Build Type: MinSizeRel
-- PX4 VERSION: v1.6.2-119-gd5eab23
-- CONFIG: nuttx-px4cannode-v1-default
-- The ASM compiler identification is GNU
-- Found assembler: /home/yang/gcc-arm-none-eabi-5_4-2016q2/bin/arm-none-eabi-gcc
-- Found PythonInterp: /usr/bin/python (found version "2.7.6") 
-- Found PY_jinja2: /usr/lib/python2.7/dist-packages/jinja2  
-- C compiler: arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 5.4.1 20160609 (release) [ARM/embedded-5-branch revision 237715]
-- C++ compiler: arm-none-eabi-g++ (GNU Tools for ARM Embedded Processors) 5.4.1 20160609 (release) [ARM/embedded-5-branch revision 237715]
-- Using C++03
-- Release build type: MinSizeRel
-- Adding UAVCAN STM32 platform driver
-- Configuring done
-- Generating done
-- Build files have been written to: /home/yang/Firmware/build_px4cannode-v1_default
Scanning dependencies of target git_genmsg
Scanning dependencies of target git_nuttx
Scanning dependencies of target build_uavcan_bl_px4cannode-v1
make[3]: *** No rule to make target `src/firmware/nuttx/firmware_nuttx.bin', needed by `org.pixhawk.px4cannode-v1-1.0-0.1.d5eab23f.uavcan.bin'.  Stop.
make[2]: *** [CMakeFiles/build_uavcan_bl_px4cannode-v1.dir/all] Error 2
make[2]: *** Waiting for unfinished jobs....
Scanning dependencies of target git_gencpp
[  1%] [  1%] [  2%] Generating git_init_Tools_genmsg.stamp
Generating git_init_NuttX.stamp
Generating git_init_Tools_gencpp.stamp
[  2%] [  2%] Built target git_genmsg
Built target git_nuttx
[  2%] Built target git_gencpp
make[1]: *** [all] Error 2
make: *** [px4cannode-v1_default] Error 2
yang@yang-B85M-D2V:~/Firmware$

davids5, 2017-06-28:
@leomatrix please install nija AKA nija-build
Then use `make clean` and `make px4cannode-v1_default`
@dagar FYI

dagar, 2017-06-28:
Well that's bizarre. Let's make sure both ninja and make fully work in https://github.com/PX4/Firmware/pull/7397