PX4/PX4-Autopilot #7488: PX4 1.6.1: Brutal oscillation in Position mode.

tubeme, 2017-06-27:
Hello guys,

We are trying our VTOL only in quad mode with the 1.6.1. The machine is flying very well in Manual and Altitude modes. But switching to Position was accompanied by very very nasty osculations. I'm clueless of what might cause the problem. Having hard time to analyze the ULOG with Flight Plot (I'm used to the old LOG format).

I noticed that there is huge jump in the Vel.X SP but cannot find the reason for it:
[image]

Here is the log:
http://review.px4.io/plot_app?log=936c8873-10f8-43c6-ad79-0faa1e98299f

Any Ideas?

tubeme, 2017-06-27:
Also I noticed a couple of messages in Flight plot around the time of the Position switching:

[image]

dagar, 2017-06-27:
It's an EKF issue, although those ekf2 low on stack messages are an unrelated warning. The important ones are "EKF GPS fusion timeout".

At a glance it's something to do with your mag.

[image]

lamping7, 2017-06-27:
I'm not sure if this is helpful, but we saw the same issue in our logs today running v1.6.3 (px4fmu-v3) on a Pixhawk 2.1 w/ HERE GNSS. We were performing some tuning flights for our MC in Stabilized mode, so there was no effect on flight, but similar errors were present. Here are links to 2 logs that have the "EKF GPS fusion timeout" warning. 
http://review.px4.io/plot_app?log=6007ff57-aa44-4631-9dfe-c7cdec8af339
http://review.px4.io/plot_app?log=edb665d3-fb10-4f93-b990-974731c852ab

Our teams next step is to fly in AUTO/Position modes, so figuring out the cause of this is of great importance.

I'm also seeing some logging dropouts, but that is a separate issue.

dagar, 2017-06-27:
@lamping7 there are position and velocity innovation test failures in both of your logs, but it's not yet clear to me why. At a minimum set EKF2_GPS_DELAY to 110 if you're using a ublox m8.
[image]

lamping7, 2017-06-28:
@dagar Thanks for the tip! I'll give that a shot as the HERE has a M8N chip. I'm not entirely sure how the EKF2_GPS_DELAY and EKF2_ASP_DELAY params became non-default, but I'll adjust and see if I can replicate the issues. Much appreciated.

tubeme, 2017-06-28:
Can we apply the same delays as a test? Ours are 200.

lamping7, 2017-07-04:
Did some more flying and the error bit didn't set. Thanks again.

tubeme, 2017-07-06:
It is better with the new defaults. We changed the GPS as well, so it is working now.

closed 2017-07-06 by tubeme