PX4/PX4-Autopilot #7489: ULOG Flight Plot: Ground Speed

tubeme, 2017-06-27:
Sorry for the silly question but I really cannot find a way to see the ground speed in Flight Plot with the new ULOG format.

We are doing MS5525 tests and I need the ground speed to compare it to the airspeed and I'm struggling in Flight Plot.

Can you point me to the parameter I should load in Flight Plot for Ground Speed?

Thanks

dagar, 2017-06-27:
In FlightPlot use the velocities from vehicle_global_position and Abs.

[image]

If you have suggestions for flight review you can open issues here. https://github.com/PX4/flight_review

closed 2017-06-27 by dagar

tubeme, 2017-06-28:
@dagar Thanks!