PX4/PX4-Autopilot #7490: Servo vibration after update

JonReacher, 2017-06-27:
After updating firmware through Qgroundcontrol I am experiencing severe strain/ vibrations in my elevon servos. After testing older firmwares and Ardupilot with no issues, it is only the PX4 firmware update causing this. Is there a known issue that I'm not aware of, or a change that needs to be made on my end to eliminate this?

dagar, 2017-06-27:
Can you share a log? Under what circumstances is this happening?

JonReacher, 2017-06-27:
I have a plane set up with the tbs caipirinha airframe preset. I've been using it for 6 months. It's been running 1.5.5. I updated yesterday to 1.6.3 through Qgroundcontrol firmware update and the servos immediately started grinding/vibrating/ random movements. They respond to manual input but with lag and they are still under strain the entire time, armed or not.

 I took the pixhawk out of the loop and they were fine under rc control. I switched to ardupilot firmware and they were fine. I switched back to 1.6.3 and it began having the same issues. I tried different flying wing presets and they all suffered the same. I tried setting pwm parameters to all default. No change.

JonReacher, 2017-06-27:
Can you use a .px4log?

dagar, 2017-06-27:
Logs before and after would be helpful.

JonReacher, 2017-06-27:
Here's an after: http://logs.px4.io/plot_app?log=0cce62f1-a27b-4350-afcd-293e33fd9e6b

I'll upload a before now.

JonReacher, 2017-06-27:
I should mention that it was any position on the main outputs that it occurred.

dagar, 2017-06-27:
Is this reproducible with bench testing or are you only seeing it in flight?

dagar, 2017-06-27:
I can compare your logs later, but as a quick thing can you first fully recalibrate all sensors with the new firmware?

[image]

JonReacher, 2017-06-27:
Here's a plot of a flight before: http://logs.uaventure.com/view/zQfdzTjo2BVdyDDDxUZeJW#top

This was only on the bench. I would not have been able to fly in this condition. The servos were getting hot under the strain.

JonReacher, 2017-06-27:
Yeah, I did full recalibration 20 times yesterday. That log was I guess after a fresh reflash, before calibration. I'll reflash, recalibrate and create a more controlled log.

JonReacher, 2017-06-27:
One thing I notice is that on the old log I see the expected actuator channels, 0,1,2, and 3. On the new log I only see outputs 2,3, and 7.

JonReacher, 2017-06-27:
Here's a log from yesterday with calibration: http://logs.px4.io/plot_app?log=7d35482b-958d-4e45-9add-f589d62f61ea

dagar, 2017-06-27:
We need to find the disconnect between what you're experiencing and what the log shows.
Here's the PWM outputs from your first log.

[image]

dagar, 2017-06-27:
PWM outputs again in the 2nd log. 

[image]

dagar, 2017-06-27:
Can you try the following to see if it makes a difference?

PWM_MIN 900 -> 1000
PWM_MAX 2100 -> 2000
PWM_RATE 400 -> 50

Have you ever connected a console? https://dev.px4.io/en/debug/system_console.html
If you can, run the command `pwm info`.

JonReacher, 2017-06-27:
Alright, I did a new flash to 1.6.3, full calibration, and created 2 logs, before and after your recommended changes. The changes did work. 
 Here is the before: http://logs.px4.io/plot_app?log=b5f158f0-c9c4-4111-9182-48a86f0c810c
 Here is after the changes: http://logs.px4.io/plot_app?log=24cc86e0-0d6b-48c4-ae45-1a4d7ed579fd

JonReacher, 2017-06-27:
I changed just the PWM_RATE back to the default of 400 and it has the issues again, so that must be it. That's probably the only relevant parameter that I didn't change yesterday.

mhkabir, 2017-06-27:
Why were you running `PWM_RATE` of 400 Hz on a plane in the first place? Standard analog servos only 50 Hz.

JonReacher, 2017-06-27:
I applied the tbs caipirinha settings to a similar airframe. I've been using this setup for a while now with no issues. I have no idea why the rate was 400HZ. I had never looked at that setting in the parameters before. Is there a reason why it would be set at that default?

JonReacher, 2017-06-27:
To me the question remains why would a 400HZ setting act differently between versions?

mhkabir, 2017-06-27:
There was a bug where the 400Hz rate wouldn't actually get applied before. It was fixed in v1.6 with the FMU app and PWM refactoring.

JonReacher, 2017-06-27:
Alright, that answers it then. Is there a way to default the rate for different classes of vehicles?

dagar, 2017-06-27:
@mhkabir the rate defaults are a bit of a mess and I think it slipped through in the past due to the rate group bug.
We need to do a pass and make sure everything is actually working as intended.
https://github.com/PX4/Firmware/issues/7261

closed 2017-06-27 by JonReacher

mhkabir, 2017-06-27:
@dagar Yeah, agreed.