PX4/PX4-Autopilot #7491: [EKF2/Vision] huge local position drift after vision interrupt

TobiMiller, 2017-06-27:
Hey,

i have switched from LPE to EKF2 for my vision input. No struggle with waiting for "Vision position init" anymore.

EKF2_AID_MASK = 24 // vision position fusion + vision yaw fusion
EKF2_HGT_MODE = Vision

I fuse the vision (working pretty well) and stop the transmission. after a little while my local position is x>>1000m and y>>1000m. if i start my fusion again, it doesnt return to the values i fuse.

is this position estimation based on acceleration and can i disable it? Is it a bug or a feature i dont understand correctly?

best regards

eric1221bday, 2017-06-29:
I actually ran into this issue before as well, though what I think is the correct behavior is that the EKF is suppose to zero the position estimation if nothing is getting through. Most of the time I do in fact get zeros if I switch vision off, though sometimes I get the extreme drift as stated above

priseborough, 2017-06-30:
Please provide a log gathered with EKF2_REC_RPL=1 and SDLOG_MODE=1 so this can be reproduced and fixed on replay.

TobiMiller, 2017-07-03:
@priseborough 
Here is the flightlog: http://logs.px4.io/plot_app?log=d24243e7-abe1-4be5-9c30-addd12d7b63f

took some time to get this values. it begins at 13:20. i fuse only vision. you can see that something goes crazy. velocity above 500 m/s.

the actuactor is also doing something crazy. the UAV stood on ground from 6.40 until end,

priseborough, 2017-07-04:
I have reviewed the log and code and the external vision fusion does not have the timeout and reversion to non-aiding mode implemented like the other observation types (GPS, optical flow) do. I will work up a PR for that.

Fortunately the log does replay.

priseborough, 2017-07-04:
@TobiMiller can you please test the patches in this PR https://github.com/PX4/ecl/pull/293 and let me know if it solves your problem.

TobiMiller, 2017-07-06:
@priseborough please look into https://github.com/PX4/ecl/pull/293 comment

TobiMiller, 2017-07-16:
fixed! can be closed.

mhkabir, 2017-07-16:
Reopening. Let's keep this open till the ECL submodule is updated.

closed 2017-08-13 by mhkabir