PX4/PX4-Autopilot #7492: MODE: Unsupported FCU while testing ros package

anudeepsekhar, 2017-06-27:
im testing https://github.com/SujithSizon/ifor_o... on my drone with pixhawk connected via ftdi cable

on qgc sys_companion was set to 921600

here is my launch file

```
<launch>

<arg name="fcu_url" default= "/dev/ttyUSB0:921600"/>
<arg name="gcs_url" default="udp://:14556@192.168.1.5:14550" />
<arg name="tgt_system" default="1" />
<arg name="tgt_component" default="1" />

<node name="mavros" pkg="mavros" type="mavros_node" output="screen" required="true" clear_params="true">
	<param name="fcu_url" value="$(arg fcu_url)" />
	<param name="gcs_url" value="$(arg gcs_url)" />
	<param name="target_system_id" value="$(arg tgt_system)" />
	<param name="target_component_id" value="$(arg tgt_component)" />
	<param name="conn_heartbeat" value="5.0" />
	<param name="conn_timeout" value="5.0" />
	<param name="startup_px4_usb_quirk" value="true" />
	<param name="mocap/use_tf" value="true"/>
	<param name="mocap/use_pose" value="false"/>

	<rosparam command="load" file="$(find mavros)/launch/px4_pluginlists.yaml" />
	<rosparam command="load" file="$(find mavros)/launch/px4_config.yaml" />
</node>

<!--node name="ifor_offb2_node" pkg="ifor_offb2" type="ifor_offb2_node" output="screen">
</node-->

</launch>
```

here is the error output

> serf@serf-ubbu:~/catkin_ifor_ws$ roslaunch ifor_offb2 ifor_offb2_sim.launch
> ... logging to /home/sujith/.ros/log/466c714a-f7ac-11e6-a85a-201a069ba36f/roslaunch-sujith-ubbu-4366.log
> Checking log directory for disk usage. This may take awhile.
> Press Ctrl-C to interrupt
> Done checking log file disk usage. Usage is <1GB.
> 
> started roslaunch server http://sujith-ubbu:34754/
> 
> SUMMARY
> ========
> 
> CLEAR PARAMETERS
>  * /mavros/
> 
> PARAMETERS
>  * /mavros/cmd/use_comp_id_system_control: False
>  * /mavros/conn/heartbeat_rate: 1.0
>  * /mavros/conn/system_time_rate: 1.0
>  * /mavros/conn/timeout: 10.0
>  * /mavros/conn/timesync_rate: 10.0
>  * /mavros/conn_heartbeat: 5.0
>  * /mavros/conn_timeout: 5.0
>  * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
>  * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
>  * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
>  * /mavros/distance_sensor/hrlv_ez4_pub/orientation: ROLL_180
>  * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
>  * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
>  * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
>  * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
>  * /mavros/distance_sensor/laser_1_sub/id: 3
>  * /mavros/distance_sensor/laser_1_sub/orientation: ROLL_180
>  * /mavros/distance_sensor/laser_1_sub/subscriber: True
[... 105 lines cut ...]
> [ INFO] [1487623163.303814321]: Plugin px4flow loaded and initialized
> [ INFO] [1487623163.308999712]: Plugin rc_io loaded and initialized
> [ INFO] [1487623163.309165119]: Plugin safety_area blacklisted
> [ INFO] [1487623163.315203583]: Plugin setpoint_accel loaded and initialized
> [ INFO] [1487623163.330421777]: Plugin setpoint_attitude loaded and initialized
> [ INFO] [1487623163.337432637]: Plugin setpoint_position loaded and initialized
> [ INFO] [1487623163.351500556]: Plugin setpoint_raw loaded and initialized
> [ INFO] [1487623163.354667402]: Plugin setpoint_velocity loaded and initialized
> [ INFO] [1487623163.361353356]: Plugin sys_status loaded and initialized
> [ INFO] [1487623163.365031065]: Plugin sys_time loaded and initialized
> [ INFO] [1487623163.366439474]: Plugin vfr_hud loaded and initialized
> [ INFO] [1487623163.366496505]: Plugin vibration blacklisted
> [ INFO] [1487623163.378667528]: Plugin vision_pose_estimate loaded and initialized
> [ INFO] [1487623163.382559340]: Plugin vision_speed_estimate loaded and initialized
> [ INFO] [1487623163.386606565]: Plugin waypoint loaded and initialized
> [ INFO] [1487623163.386680579]: Autostarting mavlink via USB on PX4
> [ INFO] [1487623163.386739021]: Built-in SIMD instructions: SSE, SSE2
> [ INFO] [1487623163.386777317]: Built-in MAVLink package version: 2016.5.20
> [ INFO] [1487623163.386798265]: Built-in MAVLink dialect: ardupilotmega
> [ INFO] [1487623163.386814579]: MAVROS started. MY ID 1.240, TARGET ID 1.1
> [ERROR] [1487623169.335451643]: MODE: Unsupported FCU

Also tried commenting out arming and setmode code and did rosrun mavros mavsis mode -c OFFBOARD

which again resulted in [Error] ......... MODE: Unsupported FCU

TobiMiller, 2017-07-03:
are you using the PX4 Stack? what happens when you start 
`roslaunch mavros px4.launch`
?

TSC21, 2017-07-03:
@anudeepsekhar can you please paste the output of `rostopic echo -n1 /diagnostics` and `rosrun mavros checkid` after you run the node? From what usually happens, `[Error] ......... MODE: Unsupported FCU` may appear after a bad connection, so I'm asking to show the output of `diagnostics` to be sure. If that's the case, may be bad wiring or faulty FTDI module/cable.

TSC21, 2017-07-07:
@anudeepsekhar updates?

anudeepsekhar, 2017-07-07:
@TSC21 sorry for the late reply i will paste the output of the diagnostics by tomorrow morning as i am currently away from my lab.

anudeepsekhar, 2017-07-09:
@TSC21 When we checked the connection and re soldered the connections the off-board was working and the issue is solved now. It was a wiring issue.

closed 2017-07-09 by anudeepsekhar