PX4/PX4-Autopilot #7495: Pixhawk Pro (V4Pro) Quad near flyaway

santiago3dr, 2017-06-27:
Took off in position mode
Almost immediately after taking off, copter started to fly away and would not respond to user input
eventually regained control in altitude mode after several minutes and several hundred meters away
http://review.px4.io/plot_app?log=2b1d4ff4-e15e-48ac-870d-2ebbe8748afc
[image]

Two succesful flights using same setup, same code, same flightplan loaded:
http://review.px4.io/plot_app?log=776e6bbf-2971-4195-8e22-6b9d0ffe167e
http://review.px4.io/plot_app?log=2b3f59bb-42d5-43b6-aaea-2e335318e275

Had another near flyaway on this same copter a few days ago:
http://review.px4.io/plot_app?log=00b0b048-39bd-45cd-910b-c3d54b0f39ad

Copter consists of
Pixhawk Pro
3DR ublox M8N with HMC5983 magnetometer
DJI F450 quad frame
DJI E300 propulsion system
3S 5200mAh battery
FrSky D4R-ii receiver

@priseborough @bkueng @dagar

LorenzMeier, 2017-06-29:
@bkueng @priseborough Can you have a look at this again?

priseborough, 2017-06-29:
OK, the 2b1d4ff4 log is the one I looked at previously with the large 180ms timeslip at start of flight and bad quality magnetometer data.

[image]

The 00b0b048 log also has a significant timeslip (not as large) and poor quality mag data

[image]

The effect of time slips depends on the angular motion of the copter at the time, If the copter generates an transient angular rate at takeoff of 100 deg/sec and we have a 100msec time slip, then that equates to 10 deg angular error, which if it occurs in roll or pitch will cause the inertial nav to accelerate and reject the GPS. This appears to be what has happened with some of these logs.

Log  776e6bbf experienced a severe time slip mid-flight, but was fortunate that it did not result in large inertial nav errors:
 
[image]

Log 2b3f59bb has one at the start of flight:

[image]

bkueng, 2017-06-30:
I see this in some of the logs:
```
[uorb] node_open as advertiser failed.
[logger] Ack timeout. Stopping mavlink log
```
around the same time where time slips happen. But not in all of them. I was able to reproduce the errors once with ulog streaming, but it does not happen always. I'll need to investigate further.

@santiago3dr I suggest you disable ulog streaming in QGC (https://dev.px4.io/en/log/logging.html#log-streaming). In recent QGC versions this is the default.
Do you use a 3DR telemetry link to connect to QGC?

santiago3dr, 2017-06-30:
yes, we use telemetry and QGC on a mac when flying the pixhawk pro
will make sure to disable log streaming

LorenzMeier, 2017-07-31:
Another log that seemed similar: http://review.px4.io/plot_app?log=c8e91a76-2481-4449-9b7e-919a21671878

TSC21, 2017-08-20:
@LorenzMeier @priseborough should this be addressed in the next days?

bkueng, 2017-08-21:
I fixed the logger-related issues in https://github.com/PX4/Firmware/pull/7783, but I did not see how it could have affected the flight performance (either through memory corruption, busy loops or blocking tasks).

mrpollo, 2017-10-05:
@PX4TestFlights Can you please increase testing on V4Pro?, we would like to understand the current state and understand what priority this should have for the 1.7 release

mrpollo, 2017-10-05:
V4Pro looks solid, here are some recent logs

https://logs.px4.io/plot_app?log=0334500c-cdaa-4c33-8b25-effac2daa066
https://logs.px4.io/plot_app?log=49562c7b-21dd-4a4e-95be-ec73ba022cd6
https://logs.px4.io/plot_app?log=d494e071-0fbf-44b4-b0da-87e23fe88f71
https://logs.px4.io/plot_app?log=97c0cc8b-6543-4c3c-bf49-5bc865b07d19
https://logs.px4.io/plot_app?log=0a9a317e-5211-4dc3-b4f2-99c8e57e74cb
https://logs.px4.io/plot_app?log=01b3edde-527a-41f8-80f3-82dd4b9ed28e
https://logs.px4.io/plot_app?log=d7d92287-ade2-4ace-86fa-7d7dc1a0c5d5
https://logs.px4.io/plot_app?log=8d2042e2-1cb6-407c-bc2f-ae483ac209d7
https://logs.px4.io/plot_app?log=f23f0250-226b-4e80-9ad7-0cb68b2c23c9
https://logs.px4.io/plot_app?log=cabdf8d8-749a-4bc2-8f42-600ac21b41e9

bkueng, 2017-10-05:
Thanks @mrpollo, I can also confirm that the V4Pro generally works well, as I have a vehicle with it and fly it more or less regularly on master.
Closing this then.

closed 2017-10-05 by bkueng