PX4/PX4-Autopilot #7501: Unexpected behavior on data-link-lost RTL

superware, 2017-06-28:
Hello,

PX4 1.6.2, flight description:

1. Arm + takeoff
2. Reposition to some point ~1 km away.
3. After 110 m - data link lost.
4. Immediate acceleration at the same flight heading (opposite to home), while gaining RTL altitude.
5. Stopping after ~160 m, and flying backwards to the point of the link-lost.
6. Rotating to home position, returning and landing.

This immediate acceleration to high speed is highly unexpected and dangerous. This has first happened live, and then reproduced with SITL.

Please see http://review.px4.io/plot_app?log=aa1a9818-dcc8-4ddc-8d1b-397117c05c4c

Thank you in advance.

nkhoit, 2017-06-28:
@Stifael @MaEtUgR This is related to #7467 
[image]
On the switch from "Hold" to "RTL" a huge velocity x gets through.

MaEtUgR, 2017-07-05:
@nkhoit indeed
@superware Thanks for the report! We track this issue on https://github.com/PX4/Firmware/issues/7467 to not have duplicate threads.

closed 2017-07-05 by MaEtUgR