PX4/PX4-Autopilot #7510: [L1 controller] Fly away because of NaNs

bresch, 2017-06-29:
Last day I had a fly away with my VTOL doing a mission. Fortunately, I was able to take it over the RC and come back in manual. 
Inspection of the log revealed that the L1 controller was outputting NaNs for roll and yaw setpoints.

Here is the log: http://logs.px4.io/plot_app?log=d72b30cd-d713-4c43-9f32-30510657f2e6

And here is a dump of the setpoints getting into a NaN state:
[image]

LorenzMeier, 2017-06-29:
@bresch that was not the L1 controller. I can almost guarantee that it was fed NaN setpoints. @Stifael can you pitch in?

dagar, 2017-06-29:
Yes, so far I'm seeing a NaN in current.alt when it starts.

dagar, 2017-06-29:
At ~479s - end the previous lat/lon are NaN, but valid is true.

dagar, 2017-06-29:
So ~479s is when the setpoint type switched from position to loiter. Previous shouldn't be marked valid, but FW loiter isn't using it.

[image]

The estimated position and current lat/lon all seem fine during the transition. The alt setpoint is NaN, but that isn't used to calculate the roll and yaw setpoints.

bresch, 2017-06-29:
Actually, the loiter worked. It went wrong just after that

dagar, 2017-06-29:
My bad, I flipped the colors. It's the end of the loiter back to position where it happens.

tl;dr the problem is the previous waypoint lat/lon are NaN, but it's marked valid. @Stifael

dagar, 2017-06-29:
The problem is the LOITER exit here - https://github.com/PX4/Firmware/blob/master/src/modules/navigator/mission_block.cpp#L374

The next position setpoint is the same as the current, which means the distance is 0 and the inner_angle becomes NaN.

Fix
1. Check the range here - https://github.com/PX4/Firmware/blob/master/src/modules/navigator/mission_block.cpp#L376
2. Populate position setpoint next

@NaterGator FYI

Thanks to the original mission file from @bresch I was able to reproduce the original in SITL.
http://logs.px4.io/plot_app?log=7b4aba61-9165-424f-9578-6779e8f66d98

[image]

@bkueng what do you think about dumping the original dataman mission items to logger? Then adding something to pyulog to output a .mission file? I know they're already on the sdcard, but I can think of many occasions where this would have been extremely helpful.

closed 2017-06-29 by LorenzMeier

bkueng, 2017-06-30:
> @bkueng what do you think about dumping the original dataman mission items to logger? Then adding something to pyulog to output a .mission file? I know they're already on the sdcard, but I can think of many occasions where this would have been extremely helpful.

Interesting. I agree this can be helpful. I see some difficulties with an actual implementation though:
- reading from dataman (SD card) will block, which can lead to logging dropouts. It can be reduced by splitting the reads over multiple loop iterations, but might still be an issue, and can take a while for larger missions.
- a mission can be updated in-air, so there could be multiple missions in one log.
-  Then we would also need some standardized format so that changes to the dataman structs won't break the parsing in pyulog.

NaterGator, 2017-07-07:
Sorry just catching up on this after getting back from Canada. Out of curiosity what was in the mission file? Was there intent to have a tangent crosstrack exit from the loiter or was that also a bug?

dagar, 2017-07-07:
It was just a regular timed loiter midmission. I believe the loiter was added in QGC without changing any defaults.

ToppingXu, 2019-09-30:
I had a same flight issue,  but the code is with this fixing. Is there any further fixing about this issue?

julianoes, 2019-10-01:
@ToppingXu please create a new issue and add a logfile.

ToppingXu, 2019-10-09:
@julianoes now I have found the root reason.
https://github.com/PX4/Firmware/blob/master/src/modules/navigator/mission_block.cpp#L390
The code above will return a nan inner angle when the loitor radius is bigger than the range.

julianoes, 2019-10-09:
@ToppingXu good catch! Do you have an idea how it could be fixed? :thinking:

julianoes, 2019-10-09:
@ToppingXu I've made a suggestion in #13138.

ToppingXu, 2019-10-10:
@julianoes good job. one more question, I am not sure whether below issue happen.
float a = 1.0f;
float b = a;
if (b > 1.0f) {
   is there any possible code run here due to the float precise
}

julianoes, 2019-10-10:
@ToppingXu I would assume that `1.0f > 1.0f` is never true for floating point. I don't see how it would be a different 1.0f :smile:.

PR #7514 (2017-06-29): navigator mission_block check range for loiter exit
 - fixes #7510 

I haven't figured out why next isn't populated, but this prevents the flyaway.

Tested in SITL. http://logs.px4.io/plot_app?log=2b554775-45c8-43c0-8be4-15d2ef9ff3bf

The MC yaw after VTOL back transition is wrong, but we can look at that later. FYI @sanderux
commit c8189b9c (2017-06-29): navigator mission_block check range for loiter exit

 - fixes #7510
PR #13138 (2019-10-09): navigator: bugfix to prevent NaN setpoints
**Describe problem solved by the proposed pull request**
The asinf function is NaN outside of -1 to 1. Therefore, it probably makes sense to constrain the input to prevent NaN setpoints further down the line.

**Test data / coverage**
It compiles :man_shrugging:.

Could potentially fix #7510.