PX4/PX4-Autopilot #7511: rc calibration arming still possible

Stifael, 2017-06-29:
During normal RC calibration, the vehicle still goes through all arming checks. This results in having the vehicle being armed and the motors running during calibration.

dagar, 2017-06-29:
I'm wondering why I've never seen this. Is it just about holding it long enough?
What board is this and do you have the safety button disabled? What's COM_RC_IN_MODE set to?

bkueng, 2017-06-29:
This happened to us on a pixhawk 3 pro with safety button disabled, `COM_RC_IN_MODE` is default=0. It probably only happens if you already have a valid channel assignment, i.e. already did an RC calibration before.

dagar, 2017-06-29:
Obviously we want to explicitly prevent arming during calibration, but should we also consider changing the arming via sticks procedure? Require both sticks moved to the same corner? Then we wouldn't need to treat it differently for FW & MC.

LorenzMeier, 2017-06-29:
No, still not safe for fixed wing. The GCS also sends a command to disable sticks during calibration. It just needs to be fixed.

closed 2018-08-01 by santiago3dr