PX4/PX4-Autopilot #7516: jMavSim: Quadrotor never takesoff with latest v1.6.3 Stable Release

sytelus, 2017-07-01:
I think something has happened after 1.0.6 RC1 release. In jMavSim, I can arm/disarm quadrotor but it never takesoff from the ground. Sometime I see message "takeoff detected" even though quadrotor is still on the ground and not moving. I just tested out 1.6.3 stable release and can reproduce issue consistently on two Pixhawk devices. The issue doesn't occur in 1.0.6 RC1 release. I haven't tried other releases.

sytelus, 2017-07-27:
Hi, any update on this? There are several users in AirSim is trying out PX4 and hitting wall with this issue. It looks like vehicle is quite shaky in last working release 1.5.5/

bkueng, 2017-07-27:
Hi @sytelus. Can you tell us exactly how you start HIL, including airframe config etc? I just used HIL on a Pixracer with master and jMAVSim a few days ago and it works fine.

LorenzMeier, 2017-07-29:
I verified it also here - working perfectly fine with jMAVSim on Mac OS.

[image]

I'm currently working on code and docs to make it easier to set up.
https://github.com/PX4/Firmware/pull/7697

closed 2017-07-29 by LorenzMeier