PX4/PX4-Autopilot #7519: PX4, EKF Time Accuracy

nicolaerosia, 2017-07-03:
Hello,

- Is there a particular reason why `hrt_abstime` (it's `uint64_t`) is measured in us and not ns ? If the accuracy on STM32 is at us level, we could just multiply and still use ns.
- Integral_dt for gyro/accel are `float32` and expressed in seconds, can we use int32_t and nanoseconds to avoid rounding off ? EKF accumulates timeslip because of this. (@priseborough)
- On POSIX, a mutex is acquired and released on every timestamping call but the mutex is only used in SITL, is it acceptable to enable it only on CONFIG_ARCH_BOARD_SITL ?

I'm willing to work and upstream this if you agree.

mhkabir, 2017-07-04:
PX4 uses microseconds for everything because of the STM32 hardware it originated from, but do you really see an advantage of having nanosecond resolution for most flight control tasks? And in any case, transitioning the codebase from us to ns would be non-trivial because of the sheer size of the change.

> Integral_dt for gyro/accel are float32 and expressed in seconds, can we use int32_t and nanoseconds to avoid rounding off.

This makes a lot of sense to me.

priseborough, 2017-07-04:
The error this rounding error introduces is negligible for a 1kHz sampling as it equates to less than 0.1% error which compared to our gyros which can have tup to 3% scale factor error, is insignificant. However if we start pushing sampling up to 4 or 8kHz to reduce aliasing effects, then it will become an issue because the effect of rounding error scales with sampling frequency.

nicolaerosia, 2017-07-04:
@mhkabir Indeed, timestamp is being used in a lot of modules which would need review. As a first step, we could migrate the deltas from messages to ns. In any case, I don't expect it to be too hard because the timestamp is usually used for comparisons and for some modules which don't need ns accuracy, we could just scale down to micros.

bkueng, 2017-07-04:
> @mhkabir Indeed, timestamp is being used in a lot of modules which would need review

The implications are much bigger: the us timestamp is embedded in uorb, and this data is written to the logs, and then used by Flight Review, FlightPlot, etc.

mhkabir, 2017-07-04:
Agree with @bkueng. I do not think that it's a good idea to change the timestamps to `ns` at this point.

closed 2018-02-21 by PX4BuildBot