PX4/PX4-Autopilot #7522: Build does not work on Windows

hamishwillee, 2017-07-04:
I followed the [Windows build instructions](https://dev.px4.io/en/setup/dev_env_windows.html) and then tried to build jmavsim simulator as per [Building the code](https://dev.px4.io/en/setup/building_px4.html)

First issue was cmake not found warning in PX4 console. Fixed by installing the 64bit Cmake for windows from [here](https://cmake.org/download/)
- Note to self, add this as dependency to docs :-)

Now trying this and getting warning that jinja2 is missing. 
- Should this be present in the "PX4 Toolchain Installer v14 for Windows Download" tools? (ie is error due to missing from tools or some weird configuration)

I did a `pip list` in the console and got my normal Windows pip2 info returned - so IF it comes with the installer it isn't being found. I then tried to add it using `pip install jinja2` which appears to have succeeded BUT which does not get detected.

```
hamis@DESKTOP-VSM2MAH /d/Github/forks/px4/Firmware
$ make posix jmavsim
-- CMAKE_MODULE_PATH: D:/Github/forks/px4/Firmware/cmake
-- Build Type: RelWithDebInfo
-- PX4 VERSION: v1.6.2-139-gf54a6c2
-- CONFIG: posix-sitl-default
-- CMAKE_INSTALL_PREFIX: /usr
-- The CXX compiler identification is GNU 4.6.2
-- The C compiler identification is GNU 4.6.2
-- The ASM compiler identification is GNU
-- Found assembler: C:/px4/toolchain/bin/gcc.exe
-- Check for working CXX compiler: C:/px4/toolchain/bin/g++.exe
-- Check for working CXX compiler: C:/px4/toolchain/bin/g++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: C:/px4/toolchain/bin/gcc.exe
-- Check for working C compiler: C:/px4/toolchain/bin/gcc.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Found PythonInterp: C:/px4/Python27/python.exe (found version "2.7.3")
CMake Error at C:/Program Files/CMake/share/cmake-3.8/Modules/FindPackageHandleS
tandardArgs.cmake:137 (message):
  couldn't find python module jinja2:

  for debian systems try:

        sudo apt-get install python-jinja2

  or for all other OSs/debian:

        sudo -H pip install jinja2

   (missing:  PY_JINJA2)
Call Stack (most recent call first):
  C:/Program Files/CMake/share/cmake-3.8/Modules/FindPackageHandleStandardArgs.c
make:377 (_FPHSA_FAILURE_MESSAGE)
  cmake/common/px4_base.cmake:621 (find_package_handle_standard_args)
  CMakeLists.txt:295 (px4_find_python_module)

-- Configuring incomplete, errors occurred!
See also "D:/Github/forks/px4/Firmware/build_posix_sitl_default/CMakeFiles/CMake
Output.log".
rm: cannot remove directory `/d/Github/forks/px4/Firmware/build_posix_sitl_defau
lt': Permission denied
make: *** [posix_sitl_default] Error 1
```

Advice welcome. My path is 
```
$ echo $PATH
/bin:/usr/bin:/usr/local/bin:/c/windows/system32:/c/px4/Python27:/c/px4/Python27
/Scripts:/c/px4/toolchain/bin:.:/usr/local/bin:/mingw/bin:/bin:/c/Python27/:/c/P
ython27/Scripts:/c/ProgramData/Oracle/Java/javapath:/c/Program Files (x86)/Intel
/iCLS Client/:/c/Program Files/Intel/iCLS Client/:/c/Windows/system32:/c/Windows
:/c/Windows/System32/Wbem:/c/Windows/System32/WindowsPowerShell/v1.0/:/c/Program
 Files (x86)/NVIDIA Corporation/PhysX/Common:/c/Program Files (x86)/Calibre2/:/c
/WINDOWS/system32:/c/WINDOWS:/c/WINDOWS/System32/Wbem:/c/WINDOWS/System32/Window
sPowerShell/v1.0/:/c/Program Files (x86)/QuickTime/QTSystem/:/mingw/bin:/mingw/m
sys/1.0/bin:/c/Program Files (x86)/Intel/Intel(R) Management Engine Components/D
AL:/c/Program Files/Intel/Intel(R) Management Engine Components/DAL:/c/Program F
iles (x86)/Intel/Intel(R) Management Engine Components/IPT:/c/Program Files/Inte
l/Intel(R) Management Engine Components/IPT:/c/Program Files/Microsoft SQL Serve
r/130/Tools/Binn/:/c/Program Files (x86)/Heroku/bin:/c/Program Files (x86)/git/c
md:/c/Program Files/nodejs/:/c/Program Files (x86)/Skype/Phone/:/c/Program Files
/Git/cmd:/c/Program Files (x86)/Windows Kits/8.1/Windows Performance Toolkit/:/c
/HashiCorp/Vagrant/bin:/c/Program Files/CMake/bin:/c/Users/hamis/AppData/Local/P
rograms/Python/Python35/Scripts/:/c/Users/hamis/AppData/Local/Programs/Python/Py
thon35/:/c/Ruby22/bin:/c/Users/hamis/AppData/Roaming/Python/Scripts:/c/Users/ham
is/AppData/Local/Microsoft/WindowsApps:/c/Users/hamis/AppData/Roaming/npm

h
```

lukegluke, 2017-07-05:
> which appears to have succeeded

Are you sure that it had succeeded?
I had error `couldn't find python module jinja2`, I fix it by installing jinja2 from cmd like this:

> "d:\px4\Python27\python.exe" "d:\px4\Python27\Scripts\pip.exe" install jinja2

hamishwillee, 2017-07-05:
Thanks @lukegluke - that would work.

For the docs, I do wonder why this version of pip wouldn't be found/used by default (since the scripts dir is in the PATH). Also, was this library "supposed" to be automatically installed and that failed, or it is just a completely missing dependency.

MansyrevAY, 2018-01-25:
Have same issue on Windows 7, i have jinja2 installed, but px4 can't find it

hamishwillee, 2018-01-29:
The current windows build is pretty broken. @MaEtUgR is working on an update based on Cygwin - you can find out more about it here: https://github.com/PX4/Firmware/issues/8357

MaEtUgR, 2018-01-29:
Can we close this? I think SITL was never able to work with the original Windows toolchain before I started the Cygwin approach.

closed 2018-01-29 by MaEtUgR