PX4/PX4-Autopilot #7525: ROS/Gazebo SITL doesn't work with launch file

anuppari, 2017-07-04:
Launching the px4 SITL through ROS (e.g., `roslaunch px4 mavros_posix_sitl.launch` doesn't actually launch the px4 simulator. No error is thrown, just the following message
```
[sitl-2] process has finished cleanly
```

Also, no px4 process is running, the PX4 logo doesn't show in the terminal, and `rostopic echo -n1 /diagnostics` produces:
```
...
status: 
  - 
    level: 1
    name: mavros: FCU connection
    message: not connected
    hardware_id: udp://:14540@localhost:14557
    values: 

...
```
Running `make posix_sitl_default gazebo` seems to work fine, with the logo showing and the px4 process running.
I'm using the latest master commit 1e2ce51c6d6aa86f16368a1ad0e00fa5a48ac7f2

TSC21, 2017-07-04:
What `sitl_gazebo` version are you using? Do you have all in the same catkin ws. What MAVROS version are you using?

anuppari, 2017-07-04:
I pulled the latest px4 master, which presumably pulled the latest compatible `sitl_gazebo` master? The command `git rev-parse HEAD` results in `622efea2f29be7dc5da7494769237aacf5b625ba`

Everything is in the `$HOME/px4/Firmware` folder. I followed the build instructions on dev.px4.io.

I updated to the latest mavros through `apt-get`

TSC21, 2017-07-04:
Did you use `--recursive` flag when you cloned the repo?
I'm currently using SITL Gazebo and MAVROS without any issue so that's why that seems weird.

anuppari, 2017-07-04:
I did not, but as I said, my `sitl_gazebo` is up to date with the latest commit, I'm not getting any build errors, and all the submodules are in place.

anuppari, 2017-07-04:
I've also replicated this issue on 2 computers

anuppari, 2017-07-05:
Is there any log file or way to introspect on what is happening when it thinks the px4 has been spawned correctly?

shivangag, 2017-07-09:
I was facing the same issue on running the sitl using roslaunch. So I tried running each node in the launch file individually and surely enough the problem was with the px4 node.
On running `rosrun px4 px4`, I got this error:
```
[rosrun] You have chosen a non-unique executable, please pick one of the following:
1) /home/shivang/DevPro/Firmware/Tools/px4
2) /home/shivang/DevPro/Firmware/build_posix_sitl_default/src/firmware/posix/px4
#? 
```
The first [px4 node](https://github.com/PX4/Firmware/commit/6ea9c1877b6465f96fd2637c29ecb13b2e81ff70) is the node that is executed by the roslaunch file, instead of the actual px4 node.
I renamed the file `Tools/px4` to `Tools/px4_ver` to solve the ambiguity and the launch file seems to work fine now.

anuppari, 2017-07-10:
Yep I also ran `rosrun px4 px4` and got he same issue. [That file](https://github.com/PX4/Firmware/commit/6ea9c1877b6465f96fd2637c29ecb13b2e81ff70) is missing the `.py` extension, which is what ROS usually uses to distinguish between cpp and python nodes, hence causing the issue.

PR #7579 (2017-07-10): Fix issue #7525
Add missing `.py` extension to Tools/px4 to fix #7525
commit 1cff8602 (2017-07-10): Fix issue #7525

Add missing `.py` extension to Tools/px4 to fix #7525
closed 2017-07-10 by LorenzMeier